ChristianTremblay/BAC0 #401: Connect to existing BACnet server?

StormFires, 2023-06-19:
Hello, I am trying to make an app that would let me create virtual BACnet devices, which I managed to do just fine using some examples.  I would like to be able to use these virtual devices with a software that my company provides. This software opens a server on a local device (my pc) on the port 47808 and reads/writes in BACnet/MSTP devices, but that causes me to have difficulties using BAC0 along with that software. As a simple test I wanted to try to read into my BACnet/IP physical device and send those data to my app, but that didn't work because, well, initiating a server with ```bacnet = BAC0.connect(ip='192.168.10.200/24', port=47808)``` obviously doesn't work since the port 47808 is already being used by my company software. 

The ip adress of my physical bacnet/ip device is 192.168.10.201 and my local server's adress is 192.168.10.200

So basically I would like to be able to read into a device that is already connected to a bacnet server on the port 47808.

I found this somewhere and thought it would be helpful but it didn't seem to work:  ```bacnet = BAC0.connect(ip='192.168.10.200/24', port=47809, bbmdAddress='192.168.10.201:47808', bbmdTTL=900)```

The ideal solution would be to not have to create a server at all with BAC0, or to create a bacnet server on port 47809 with BAC0 and be able to read my device that is on port 47808, but I don't know if it's possible.

If you have any questions or would like me to provide additional information, please tell me, thank you.

bbartling, 2023-06-20:
maybe look through the closed questions...i think someone asked this recently but I think the solution was creating something with multiple network adapters on Windows OS.

I am not an expert in this area but I think the OS doesnt allow for multiple programs trying to open port 47808. Can you use multiple devices on a LAN like the typical environment for a building automation system is each operating system or device is unique all on the same network.

what was the use case again? To create BACnet server and client all on the same device?

StormFires, 2023-06-20:
Thanks for your answer @bbartling. I looked through the issues a while back and didn't find anyhting I thought I could use, but I haven't looked at them recently, so i'll do that and answer this post again if I find a solution. 

The use would be to have a BACnet server created by another software on port 47808, then create virtual devices like in this example: https://github.com/ChristianTremblay/BAC0/blob/master/tests/conftest.py , so this would indeed be on the same device as the server (my pc), and be able to communicate (read/write) with physical devices as well.

Right now, I am able to have my BACnet server from my software created on port 47808 and a BAC0 server created on another port (eg. 47809), but I can't communicate between the two. I can't even read into my physical device on port 47808 with BAC0 because BAC0 requires that port to communicate with my physical device.

You might not be familiar with node-bacnet, but it's pretty close to what BAC0 does but for javascript/node.js. With node-bacnet, I can use this code:
```
const client = new bacnet();
var type = 2
var instance = 0
var IPadress = "192.168.10.201:47808"
var value = 47.789
client.writeProperty(
    IPadress, 
    {type, instance},
    85, 
    [
    {type: bacnet.enum.ApplicationTag.REAL, value : value}
    ], 
    (err, value) => {
        
});
```
to successfully write into my physical device even though the BACnet server from my company software is opened, which I can't do with BAC0. So it is technically possible, but I don't know if you can do that with BAC0

bbartling, 2023-06-20:
ah so your using Node Red. Is Node Red your BACnet server or client?

Just shooting from the hip in BAC0 you could do both BACnet server and client on the same script. So could you run Note Red in one terminal and BAC0 in the other?

I was briefly playing around with trying to use Python BACnet stacks a while back and Node Red on another terminal and using standard in & out for communicate the data in between. I'm no expert on this stuff but maybe this would be better approach and letting one BACnet service use the 47808 port?? https://stackoverflow.com/questions/70511960/python-shell-node-keep-a-python-script-running/70520584#70520584

I found the Node Red BACnet real hard to use that was my original idea but I liked the look and feel of Node Red interface

bbartling, 2023-06-21:
feel free to post what your BACnet server and client requirements would be in an app which could be done in BAC0. what points do you need?

bbartling, 2023-06-21:
`192.168.10.201` the strato server?  I think I understand those scripts are on the same machine BAC0 is making read requests to `192.168.10.201:47808`

Can you run that by me one more time what would be the most ideal outcome? To be able to make continuous read requests with BAC0?

StormFires, 2023-06-21:
`192.168.10.201` is the BPIP Strato device which is physically connected to my computer. `192.168.10.200` is the stato server on my machine. 

My current "ideal outcome" would be to be able to read/write with BAC0 into my BPIP Strato device which is physically connected to my computer while my strato server is also active at the same time.

I can already do that with Node Red so I know it's possible in BACnet, but I would like to do that with BAC0.

My real final goal is communication between virtual BACnet devices and physical BACnet devices, while my strato BACnet server is active.

bbartling, 2023-06-21:
Is the trouble keeping your BAC0 clients scripts alive? Do they need to run
continuously on a time interval?

ChristianTremblay, 2023-06-22:
An easy way to make it work is to define more than 1 IP address for your network interface.
See [here](https://github.com/ChristianTremblay/BAC0/issues/392#issuecomment-1561932554)

192.168.10.200 + 192.168.10.202 for example...
Then strato uses 1 IP
BAC0 uses the other

Also, use `BAC0.lite` over `BAC0.connect` for those kind of things. It simpler.

For example, I have a machine configured with 2 IPs
[image]

Niagara4 uses `.5` for it's BACnet/IP port
[image]

And BAC0 uses `.4` (the script here runs as a service)
[image]

Everything on port 47808, it works

StormFires, 2023-06-22:
Thank you very much @ChristianTremblay, it works now! I didn't even know having multiple ips set up could help with this, this is very interesting.

Thank you to @bbartling as well for being patient with me and trying to help me.

closed 2023-06-22 by StormFires