ChristianTremblay/BAC0 #402: Event Triggering for BACnet Points

fpfaffendorf, 2023-06-27:
Hi there !
I couldn't answer the following question by reading the documentation.
Is there an event triggered when a BACnet point is externally modified by a client?
Thanks in advance !

bbartling, 2023-06-27:
what is your use case and desired event in your app? Any variable in Python can be triggered in your code when a point changes state from an external client writing to your app. I assume you have a BACnet server app with some points that are discoverable and are writeable by external client?

lgassman, 2023-06-28:
Hi, I am working with fpfaffendorf. We are building a BACnet device. In this device, we are setting a local point of analog value kind. The presentValue of this point could be changed due to a local event (In this case, I programmatically call a line like myObject.presentValue=Real(...)). Additionally, the present value could be changed due to a BACnet server sending a request (my analog value is commandable).

We need to identify when the presentValue is changed only due to a server request (ignoring local changes).

bbartling, 2023-06-28:
this maybe a lot of code to look at but line 68 in this app I do a recurring task every second to see if the data is updated from the BACnet server. You maybe able to do something like that run reoccurring task on a function. That should tell if the object pv changed from externally.

https://github.com/bbartling/openleadr-example-ven/blob/main/bacnet_server/main.py

bbartling, 2023-07-05:
Ive updated my method is the code example `    def update_bacnet_api(self):` to not use `recurringTask`

Chat GPT actually helped me write something where I can use asnycio and BAC0. Basically some other needs in this app is Python asnyc type code and needed a way to update a BACnet server points.

To keep the BAC0 app alive i am using:

```
    async def keep_baco_alive(self):
        counter = 0
        while True:
            counter += 1
            # update BACnet API ~ every second
            if counter == 100:
                await asyncio.to_thread(self.update_bacnet_api)
                counter = 0
            await asyncio.sleep(0.01)
```

and the BAC0 app is defined like this:

```
class BACnetApp:
    @classmethod
    async def create(cls, ven_client):
        self = BACnetApp()
        self.ven_client = ven_client
        self.bacnet = await asyncio.to_thread(BAC0.lite, deviceId=BACNET_INST_ID)
        _new_objects = self.create_objects()
        _new_objects.add_objects_to_application(self.bacnet)
        logging.info("BACnet APP Created Success!")
        return self
```

many thanks for Chat GPT it seems to work well from my testing.

Feel free to post what you are trying to do @lgassman  or @fpfaffendorf I maybe able to help some. Post some of your code as well.

lgassman, 2023-07-05:
Hi! I took a different approach. My main problem was differentiating whether a cov was produced by a local decision or a server request. I found a way to add a monitor when a cov is generated, but I kept having trouble distinguishing the source of the change.

So, I implemented a UI strategy. I envision the commandable object as a UI object and wrapped it with my own ModelObject, which keeps the actual modeled value in its 'model' attribute. It's very similar to the MVVM architecture pattern. If you have experience with frameworks like Wicket, it's the same idea.

Then, my current value is duplicated: I have a value in my model object under the 'model' attribute, and I have the same value in the commandable object under the 'presentValue' attribute. After that, I subscribe as a monitor of cov.

When I need a local change, I update the model first and then update the presentValue. My monitor checks if the values are equal, assuming the change is local and ignoring it. However, if the monitor detects different values between the model and presentValue, it assumes the change is external and fires an event.

Here's my code (I trimmed some parts):

```
class BacnetPoint(abc.ABC):
    
    def __init__(self, bacApp, instance_number, device_name,  model = None): 
        
        self.bacApp = bacApp
        self._model = model 
        self.instance = instance_number
        self.device_name=device_name
        objectFactory = self._buildWrapped() 
        objectFactory.add_objects_to_application(self.bacApp)
        self.observe_write_property()

    @property
    def model(self):
        return self._model
    
    @model.setter
    def model(self, value): 
        '''
        change the model and the component value only model is a new value. we need keep duplicated these values to detect when component is changed from bacnet network
        '''
        if self._model != value :
            self._model = value 
            self.wrapped.presentValue = value

    @property
    def wrapped(self):
        return self.bacApp[self.device_name]

    def description(self):
        return f"{self.__class__.__name__ } of {self.device_name}"
    
    
    def observe_write_property(self, property_name="presentValue"):
        if property_name in self.wrapped._property_monitors: #I am not sure if it the best way to subscribe a monitor
            self.wrapped._property_monitors[property_name].append(self.cov_monitor)
        else:
            self.wrapped._property_monitors[property_name] = [self.cov_monitor]

    def cov_monitor(self, old_value, new_value):
        if new_value != self._model:
            self._model = new_value
            self.on_model_changed_from_bacnet() 

    @abc.abstractmethod
    def _buildWrapped(self):
        raise Exception("subclass responsability")

    @abc.abstractmethod
    def on_model_changed_from_bacnet(self):
        raise Exception("subclass responsability")
    
class Dimming(BacnetPoint):

    def _buildWrapped(self):
        return analog_value (
            instance=self.instance,
            name=self.device_name,
            description=self.description(),
            presentValue=0,
            properties={
                "units":"percent",
                "minPresValue" : Real(0),
                "maxPresValue" : Real(100)
                }
            )

    def on_model_changed_from_bacnet(self):
        logging.debug("COV FROM SERVER")

```

Thanks a lot for your help.

bbartling, 2023-07-05:
nice work @lgassman. How does the code work to trigger COV? Is the scenario where something external is writing to your app point and then you can see if it changed value? How often does it sample to know if the value changed or not?

lgassman, 2023-07-06:
I am not configuring these value, I jus use default settings. Seeing logs files, apparently the cov is checked each 5 seconds.

I starts my app in this way:
```
bacApp = BAC0.connect(deviceId=12345)
Dimming(bacApp, 1, "my_ligth")
```
Maybe to change to `lite` method be better

bbartling, 2023-07-07:
@ChristianTremblay does BAC0 under the hood handle COV on a BACnet server or is that something that needs to be configured in your app code when its created something like:

```
    def create_objects(self):
        _new_objects = analog_value(
            name="demand-response-level",
            description="SIMPLE SIGNAL demand response level",
            presentValue=0,
            is_commandable=False,
        )
        _new_objects = analog_value(
            name="power-level",
            description="Writeable point for electric meter reading",
            presentValue=0,
            is_commandable=True,
        )
        _new_objects = binary_value(
            name="openadr-server-connection-status",
            description="True or False for connection status",
            presentValue=False,
            is_commandable=False,
        )
        return _new_objects
```

lgassman, 2023-07-07:
I am not doing nothing in special to use COV. I just create my bacApp using "connect" method (but I suspect to use 'lite' is the same). My app is running on a client device. In the other hand, I use a bacnet server to subscribe to COV of my device. After that I am able to change my value from client and see the change on the server (due it is subscribed to cov) and also I can send a request from server to client to change de value and see the new value on presentValue atrribute of the analog object (due I create a analog object and this is commandable).

My code only add the ability to fire an event when the value is changed from server, that was my the problem that origin this thread. We are not sure if this approach is the best way to do it but it works. We posted the quastion hoping found a better or more directly way. 

I don't knwon what is the limit between BAC0 and BACPypes. I am newby with this techonoliges.

bbartling, 2023-07-09:
@ChristianTremblay what is preferred way of building BACnet server app? What is preferred using `ObjectFactory` or just using something like `from BAC0.core.devices.local.models import analog_value, binary_value` ?

How do you import `ObjectFactory`? I cant quite see how this is done in the readthedocs:

https://bac0.readthedocs.io/en/latest/local_objects.html#an-example

Thanks!

bbartling, 2023-07-09:
@lgassman 
> I don't know what the difference is between BAC0 and BACPypes. I am a newbie to these technologies.

BAC0 runs bacpypes under the hood....and makes the development process for apps very easy

ChristianTremblay, 2023-07-10:
I need some time to read this thread....

But bottom line
As @bbartling said, BAC0 is based on BACpypes. You could achieve the same using only BACpypes, but BAC0 makes the configurtion easier... to the price of an opiniated way of doing things.

Regarding COV, it is a BACnet mechanism so 2 different BACnet devices may subscribe to each other by pointing specific objects. When this object changes the host device is allowed to talk to the COV client directly to send the new value. It is not an internal mechanism. This comes in 2 flavours : standard and confirmed. The first just yell that its value changed, the second one is more polite and we'll validate that the client knows that the value changed.

ChristianTremblay, 2023-07-11:
> Hi there ! I couldn't answer the following question by reading the documentation. Is there an event triggered when a BACnet point presented value is externally modified by a client? Thanks in advance !

@fpfaffendorf not that I'm aware of

ChristianTremblay, 2023-07-11:
> Hi, I am working with fpfaffendorf. We are building a BACnet device. In this device, we are setting a local point of analog value kind. The presentValue of this point could be changed due to a local [...]

@lgassman you could use the priority array feature. 
Internally, you could change the presentValue to whatever value using priority 16. But if the other BACnet device writes to this variable to a priority from 15 to 1, you will be able to differentiate. If the server also write to priority 16, then it would be a matter of comparing values (keeping some kind a traces).

ChristianTremblay, 2023-07-11:
> Hi! I took a different approach. My main problem was differentiating whether a cov was produced by a local decision or a server request. I found a way to add a monitor when a cov is generated, but I kept having trouble distinguishing the source of the change.
> 
> So, I implemented a UI strategy. I envision the commandable object as a UI object and wrapped it with my own ModelObject, which keeps the actual modeled value in its 'model' attribute. It's very similar to the MVVM architecture pattern. If you have experience with frameworks like Wicket, it's the same idea.
> 
> Then, my current value is duplicated: I have a value in my model object under the 'model' attribute, and I have the same value in the commandable object under the 'presentValue' attribute. After that, I subscribe as a monitor of cov.
> 
> When I need a local change, I update the model first and then update the presentValue. My monitor checks if the values are equal, assuming the change is local and ignoring it. However, if the monitor detects different values between the model and presentValue, it assumes the change is external and fires an event.
> 
> Here's my code (I trimmed some parts):
> 
> ```python
> class BacnetPoint(abc.ABC):
>     
>     def __init__(self, bacApp, instance_number, device_name,  model = None): 
>         
>         self.bacApp = bacApp
>         self._model = model 
>         self.instance = instance_number
>         self.device_name=device_name
>         objectFactory = self._buildWrapped() 
>         objectFactory.add_objects_to_application(self.bacApp)
>         self.observe_write_property()
> 
>     @property
>     def model(self):
>         return self._model
>     
>     @model.setter
>     def model(self, value): 
>         '''
>         change the model and the component value only model is a new value. we need keep duplicated these values to detect when component is changed from bacnet network
>         '''
>         if self._model != value :
>             self._model = value 
>             self.wrapped.presentValue = value
> 
>     @property
>     def wrapped(self):
>         return self.bacApp[self.device_name]
> 
>     def description(self):
>         return f"{self.__class__.__name__ } of {self.device_name}"
>     
>     
>     def observe_write_property(self, property_name="presentValue"):
>         if property_name in self.wrapped._property_monitors: #I am not sure if it the best way to subscribe a monitor
>             self.wrapped._property_monitors[property_name].append(self.cov_monitor)
>         else:
>             self.wrapped._property_monitors[property_name] = [self.cov_monitor]
> 
>     def cov_monitor(self, old_value, new_value):
>         if new_value != self._model:
>             self._model = new_value
>             self.on_model_changed_from_bacnet() 
> 
>     @abc.abstractmethod
>     def _buildWrapped(self):
>         raise Exception("subclass responsability")
> 
>     @abc.abstractmethod
>     def on_model_changed_from_bacnet(self):
>         raise Exception("subclass responsability")
>     
> class Dimming(BacnetPoint):
> 
> 
>     def _buildWrapped(self):
>         return analog_value (
>             instance=self.instance,
>             name=self.device_name,
>             description=self.description(),
>             presentValue=0,
>             properties={
>                 "units":"percent",
>                 "minPresValue" : Real(0),
>                 "maxPresValue" : Real(100)
>                 }
>             )
> 
>     def on_model_changed_from_bacnet(self):
>         logging.debug("COV FROM SERVER")
> ```
> 
> Thanks a lot for your help.

@lgassman I'm not sure relying on COV to do that is a viable solution. Not all BACnet device support COV and depending on how your thing will be used, or configured by other, by relying on COV you need to put faith on the integration of someone else.

Using the priority array is safer. The priority array mechanism roles is to give more priority to different "level" depending on the use case. Typically, internal variables or sequences are written at priority 16. The last value written being the actual value. In certain situation, your app could write to a higher priority if required.

https://store.chipkin.com/articles/bacnet-what-is-a-bacnet-priority-array

lgassman, 2023-07-11:
Thanks a lot!

fpfaffendorf, 2023-07-11:
Thanks guys, I'm closing this issue.
Best regards.

closed 2023-07-11 by fpfaffendorf