ChristianTremblay/BAC0 #405: AnalogOutput returns unrecognized-service when calling writeMultiple

sunarvin, 2023-07-05:
The presentValue and outOfService can be written individually for AO. While a unrecognized-service error is captured on Wireshark when calling writeMultiple with ip and value list.

Many thanks!

client code:
'''
#!/usr/bin/python

import BAC0

ip_addr = '192.168.1.11'

bacnet = BAC0.lite()

print('\nTry to access BACnet points:')

 ## write presentValue
bacnet.write(f'{ip_addr} analogOutput 0 presentValue 22.2')
 ## read back
print(bacnet.read(f'{ip_addr} analogOutput 0 presentValue'))

 ## write outOfService
val = bacnet.read(f'{ip_addr} analogOutput 0 outOfService')
val = not val
bacnet.write(f'{ip_addr} analogOutput 0 outOfService {val}')
print(bacnet.read(f'{ip_addr} analogOutput 0 outOfService'))

 ## write multiple value
vals = ['analogOutput 0 presentValue 33.3', 'analogOutput 0 outOfService True']
bacnet.writeMultiple(addr=ip_addr, args=vals)                       ## <== meet a unrecognized-service error here!
 ## read back
print(bacnet.readMultiple(f'{ip_addr} analogOutput 0 presentValue outOfService'))
'''

service code:
'''
#!/usr/bin/python

import BAC0
import time

from BAC0.core.devices.local.models import (
    analog_output,
)

bac_id = 666
bac_port = 47808

def start_device():
    print("Starting BACnet device")
    new_device = BAC0.lite(deviceId=bac_id, port=bac_port)
    new_device._log.info('Device ID : {}'.format(new_device.Boid))

    ## test_av
    new_objs = analog_output(
        instance = 0,
        name = "AO0",
        description = "valve",
        presentValue = 2.2,
        properties = {
            "units": "percent",
            "minPresValue": 0,
            "maxPresValue": 100,
            },
    )

    new_objs.add_objects_to_application(new_device.this_application)

    return new_device

start_device()

while True:
    time.sleep(1)
'''

ChristianTremblay, 2023-07-11:
WriteMultiple is not supported by all devices. 
This error simply tell you that the device do not support this service.

ChristianTremblay, 2023-07-11:
BTW, the sleep(1) will cause issues. Make that smaller.... 0.01 or less. 
Sleep pauses the entire python app. You will miss messages and fight very weird bugs

sunarvin, 2023-07-13:
Yes. It is confirmed that the Lite does not support the WriteMultiple feature.

`
class Lite(
    Base,
    Discover,
    ReadProperty,
    WriteProperty,
    Simulation,
    TimeSync,
    Reinitialize,
    DeviceCommunicationControl,
    CoV,
    Schedule,
    Calendar,
    TextMixin,
):`

closed 2023-07-13 by sunarvin