ChristianTremblay/BAC0 #408: AttributeError: module 'BAC0' has no attribute 'connect'

MichaelSpeckien, 2023-08-09:
I installed BAC0 as described.
When importing BAC0 I get the following message:

Import Error, refer to documentation or reinstall using
    $ "pip install BAC0"
 No module named 'pytz'

Discard this message if you are actually installing BAC0.

When calling BAC0.connect() I get the following exception:

AttributeError: module 'BAC0' has no attribute 'connect'

Can anyone help me?
BAC0.version = 23.07.02
Python 3.10.10

ChristianTremblay, 2023-08-09:
I suspect you have multiple pythoin version in your system and when you pip installed... you in fact installed BAC0 in a different version than the one your are using to import it.

Sometimes, in Linux, you should use `pip3 install BAC0`

MichaelSpeckien, 2023-08-09:
Thank you very much for the fast answer. I will remove all old files and reinstall python and the complete package.

Btw, for years I used BAC0 nearly without problems. Really a great package!!!

MichaelSpeckien, 2023-08-11:
Issue is solved.
I pip installed pytz, and after this everything worked.

closed 2023-10-18 by github-actions[bot]