ChristianTremblay/BAC0 #409: UDP port setup other than 47808

bbartling, 2023-08-25:
I have a real simple BACnet server app but I am finding out (campus setting of lots of buildings) that the organization IT department just make a flat network where then all HVAC controls contractors just have each building on a unique UDP port and a BBMD for that building.

If my BACnet server app just needs to talk to the equipment in building:
```
STATIC_IP = "10.7.6.201"
BACNET_INST_ID = 3056672
PORT = "47820"
BBMD = "10.7.6.8"
```

Is this syntax correct for BAC0 Lite app? I actually haven't ran across this type of networking scenario yet:
`device_app = BAC0.lite(ip=IP_ADDRESS, port=PORT , mask=24, deviceId=BACNET_INST_ID, bbmdAddress=BBMD)`

I can remember the bad thing to do is create 2 BBMD's on the same subnet...can anyone confirm this syntax is Okay and I wont sink the ship on a BACnet storm?

bbartling, 2023-08-25:
When would you use this type of syntax where you put everything on one line? `10.7.6.201/24:47820`

TrickPah, 2023-09-12:
Here is a discussion from 2018 with a snippet from bacpypes2.7, where the "extended default" ports 47808-47823 are coded:
https://github.com/JoelBender/bacpypes/issues/183

bbartling, 2023-09-12:
for what its worth it does work! I can run my BACnet server on port 47820 and it appears to work just fine with a BACnet scan tool discovering on port 47820.

Its some requirement for a project on a campus. Trane must have told the IT department who knows how long ago to just make the network completely flat and then each building will run a unique BACnet UDP port and a BBMD I think.

closed 2023-09-12 by bbartling