ChristianTremblay/BAC0 #411: MSTP reading

CSEGarazas, 2023-09-05:
Hello, 
I have problem with reading MSTP/IP network with Schneider controllers.
The problem is that there are a lot of IP/MSTP routers which has same network number. 
For example I have router 172.16.102.61 and 172.16.102.62 and they have same network ID 2001, Also these routers have a lot  of RoomControllers with uniq BacNetObjectIdentifier but some of them have same MAC address (2001:1 with ObjIdentifier 1283001 on router 172.16.102.62, and 2001:1 with ObjectIdentifier 1383001 on router 172.16.102.61)

I can find both RoomController using YABE, but I can`t find both them and read them using BAC0 library.
I attached code example but it gives me same value.
Also I tried to write IP address instead network:MAC but I get timeout. 

[image]
[image]
[image]

bbartling, 2023-09-05:
Oh wait maybe I understand it better that the MSTP networks are all the same and network numbers, etc...oh man it seems like a mess.

Am not really sure what you would do in this scenario. I almost think you may have to ask the controls contractor to change the MSTP network number to make them unique. Am curious what @ChristianTremblay has to say

CSEGarazas, 2023-09-05:
I tried to use BAC0.Lite and it gives same result.
Also I talked to my collegue how working with AutomatedLogic controllers and he said that it is ok that it have same Network and MAC, and most important is BacnetObjectIdentifier, but how I can reach it using BAC0 library, I don`t know :D 
I hope that some one could help me with this issue :)

ChristianTremblay, 2023-09-09:
On one BACnet network, every different network should have a unique network numbers. Clearly, re-using the same MSTP network under different routers on the same IP network... its breaking the rules.

It means that if we call `whos is router to network 2001" on the network, we'll get 2 answers, from 2 different routers... The app will not know how to deal with this.

But if your configuration looks like this instead (and the network 2001 refers to the BACnet/IP network, number, you could be ok)

[image]

Then the adresses for the MSTP devices becomes  `61:MAC` or `62:MAC`... not `2001:MAC`

Try making a `bacnet.discover()` first

CSEGarazas, 2023-09-25:
Hi, Thank you for answers.
After long time I finally found solution.
Maybe some one will help it :

`import BAC0
import datetime
from bacpypes.settings import settings
settings.route_aware=True

bacnet =BAC0.lite()

rr = bacnet.read(f'2002:8@172.16.102.61 analogValue 100 presentValue')`

bbartling, 2023-09-25:
when do people need to use? This stuff is new to me as well : )
```
from bacpypes.settings import settings
settings.route_aware=True
```

CSEGarazas, 2023-09-25:
When on same network there are several devices with same MAC address and they are on different routers (IP addresses).
When we are trying to read mstp network for example:
bacnet.read('2002:8 analogValue 100 presentValue')
Source IP address calls all subnet and asking which mstp router on network 2002, then some router answers and then BAC0 know which IP is DestinationAddress and then can call second request to current IP. 
So writing bacnet.read(f'2002:8@172.16.102.61 analogValue 100 presentValue') , we are  eliminating first request , because we already know DestinationAddress.

bbartling, 2023-09-25:
in this example of `(f'2002:8@172.16.102.61 analogValue 100 presentValue')`...the ip address of `172.16.102.61` is the BACnet router right? Then downstream of that BACnet router MSTP device hardware address 8 on network 2002

CSEGarazas, 2023-10-09:
Yes, you are right :)

mohakbhalla7, 2023-10-16:
Hi @ChristianTremblay 

I had a doubt regarding this comment of yours.
> Try making a bacnet.discover() first

I was facing a similar problem. To give a little context, multiple BACnet MS/TP devices were connected to a BACnet router, and I could not get an I-Am response from those devices. As you suggested, I was doing a `bacnet.discover()` first and then `bacnet.whois()`. The following was the piece of code

```
bacnet.discover(global_broadcast=True)
bacnet.whois()
devices = bacnet.devices
```
I was only able to get the response from the BACnet router. 

Then with a little help from #421 and this thread, I figured that just a `bacnet.whois(global_broadcast=True)` was enough, and I was able to get the response from the devices.

But when I tried a `bacnet.whois(global_broadcast=True)` after a `bacnet.discover(global_broadcast=True)`, this was the error that I was getting.
[image]

And when I tried a `bacnet.discover(global_broadcast=True)` after a `bacnet.whois(global_broadcast=True)` and tried to get the devices, this was the error that I was getting.
[image]

So this brings me to the question, of how whois() and discover() function differently than each other, if essential they are doing the same thing (doing a Who-Is broadcast request). And if they actually are different from each other, then which one should be preferred to make Who-Is requests to the devices?

ChristianTremblay, 2023-12-11:
The discover function is an attempt to be "more polite" when arriving on a network. First find the network on which we are. Then, look for routers... then for each of the routers/network send a whois.

If going with a full broadcast, you can create a "déluge" of I-Am responses that can potentially flood the network.

closed 2024-02-17 by github-actions[bot]