ChristianTremblay/BAC0 #415: Reading the Data from BACnet using Raspberry pi4

LRAJA33, 2023-09-13:
Dear All,
I am having an issue when attempting to connect my raspberry PI to BACnet. I can successfully connect my desktop to the network, but not the RPI.
I have tested with windows machine , it is working. In the machine both bacnet device ip and system ip are pinging , but in raspberry pi only Rpi ip is pinging and bacnet device ip is not pinging. Kindly help me to resolve this issue.

Code:
import BAC0
import json
import time
# create a bacnet connection to connect to the devices, use your local machine IP address and the subnet below and also specify the port number if needed
bacnet = BAC0.lite(ip="192.168.1.9/22",port=47809)  

data = {}

def main():
 
     
    id = str(1)
    # Connect to the BACnet simulator using the IP address and port number
    value = bacnet.read("192.168.1.10 analogValue "+id+" presentValue")
     
     value = str(value)
      
     data["Analog_input"+id] = value
            
     print(json.dumps(data, indent=2))
     time.sleep(5)

   bacnet.disconnect()

        
#RUN main
if __name__ == '__main__':
    main()

Terminal Output:
[image]

ChristianTremblay, 2023-09-13:
A user recently had the same kind of issue and it was the firewall blocking UDP port 47808.

TrickPah, 2023-09-13:
This console output does not fit the code provided...there is no point in which the BACnet Identifier-Tuple or a string like "Analog Input Value" is defined.

Running the code provided with a jsonified object would like so:
{
  "Analog_Input1": "<VALUE>"
}

Which will actually not print, if You fail to properly read, as that would throw an error.

LRAJA33, 2023-09-14:
Thank you for your kind reply. I will check it sir

<http://www.bitsathy.ac.in/>
Dr RAJASEKAR L | ASP
*Electronics and Instrumentation Engineering*
Bannari Amman Institute of Technology
  Sathyamangalam, Erode - 638 401 <https://goo.gl/maps/rkMMcedqLVpzaDEY9>
[image: T]  +91 96007 66788 <%20+91%2096007%2066788> | 04295 226000
<%2004295%20226000>
[image: M]  ***@***.***

closed 2023-11-20 by github-actions[bot]