ChristianTremblay/BAC0 #417: Failing to write to a point on a defined controller

dqcontrols, 2023-09-15:
Hi,

I am attempting to write to a numeric Analog Value in my controller (Edge 10) that I have commissioned in Niagara Workbench. A thermostat is attached to the RS485 ports of the Edge 10. Here is an image of the Nav Tree in Niagara of my setup. I have brought in some thermostat points and also put them in the export table to ensure they can be discovered by BAC0.
[image]

After establishing connection and defining the Edge 10 as my controller, I can read all 4 points that I have in Niagara, but I am unable to write to the Occupied Cool Setpoint, which is set up as a numeric writable in Niagara.
[image]

Here is my code:

import BAC0

# Start BAC0 Engine
bacnet = BAC0.connect(ping=True)
bacnet.discover(networks='known')
mycontroller = BAC0.device('10.0.0.92', 1000, bacnet)
properties = mycontroller.bacnet_properties
points_list = properties['objectList']
point_vals = {}
for point in points_list:
    point_vals[point] = mycontroller[point[0], point[1]]
    

Here is the console output. Note that I can read all 4 points, but cannot write to analogValue 0, which I want to set to 75F and is the thermostat occupied cool setpoint:
2023-09-15 16:09:53,977 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2023-09-15 16:09:53,981 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2023-09-15 16:09:53,982 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2023-09-15 16:09:53,985 - INFO    | Starting TaskManager
2023-09-15 16:09:54,010 - WARNING | Netifaces not installed on your system. BAC0 can't detect the subnet.
Please provide subnet for now, we'll consider 255.255.255.0 (/24).
You can install netifaces using 'pip install netifaces'.
2023-09-15 16:09:54,012 - INFO    | Using ip : 10.0.0.34
2023-09-15 16:09:54,707 - INFO    | Starting app...
2023-09-15 16:09:54,709 - INFO    | BAC0 started
2023-09-15 16:09:54,711 - INFO    | Registered as Simple BACnet/IP App
2023-09-15 16:09:54,739 - INFO    | Update Local COV Task started
2023-09-15 16:09:54,898 - INFO    | 10.0.0.92 network number is 1000
2023-09-15 16:09:56,853 - INFO    | 10.0.0.92 router to [20000]
2023-09-15 16:09:58,784 - INFO    | Addr : 10.0.0.92
2023-09-15 16:09:58,847 - INFO    | 10.0.0.92 routing table
2023-09-15 16:09:58,849 - INFO    |     1000 1 bytearray(b'Annex J IP')
2023-09-15 16:09:58,852 - INFO    |     20000 2 bytearray(b'MS/TP')
2023-09-15 16:10:00,814 - INFO    | Found those networks : {1000, 20000}
2023-09-15 16:10:00,817 - INFO    | Discovering network 1000
2023-09-15 16:10:03,848 - INFO    | Discovering network 20000
2023-09-15 16:10:06,977 - INFO    | Changing device state to DeviceDisconnected'>
2023-09-15 16:10:07,290 - INFO    | Changing device state to RPMDeviceConnected'>
2023-09-15 16:10:07,477 - INFO    | Device 1000:[Edge10_Station_1000] found... building points list
2023-09-15 16:10:07,614 - INFO    | Ready!
2023-09-15 16:10:07,616 - INFO    | Device defined for normal polling with a delay of 10sec
2023-09-15 16:10:07,619 - INFO    | Polling started, values read every 10 seconds

mycontroller['analogInput', 0]
Out[2]: Edge10_Station_1000/Drivers.BacnetNetwork.VT8600.points.Thermostat.Light Sensor Level : 303.00 noUnits

mycontroller['analogInput', 1]
Out[3]: Edge10_Station_1000/Drivers.BacnetNetwork.VT8600.points.Thermostat.Room Temperature : 23.83 degreesCelsius

mycontroller['analogInput', 2]
Out[4]: Edge10_Station_1000/Drivers.BacnetNetwork.VT8600.points.Thermostat.Room Humidity : 42.00 percentRelativeHumidity

mycontroller['analogValue', 0]
Out[5]: Edge10_Station_1000/Drivers.BacnetNetwork.VT8600.points.Thermostat.Occupied Cool Setpoint : 73.00 degreesFahrenheit

mycontroller['analogValue', 0] = 75
Traceback (most recent call last):

  Cell In[6], line 1
    mycontroller['analogValue', 0] = 75

  File ~\anaconda3\lib\site-packages\BAC0\core\devices\Device.py:624 in __setitem__
    self._findPoint(point_name)._set(value)

  File ~\anaconda3\lib\site-packages\BAC0\core\devices\Device.py:695 in _findPoint
    raise ValueError("{} doesn't exist in controller".format(name))

ValueError: ('analogValue', 0) doesn't exist in controller
    

Any help would be appreciated!

ChristianTremblay, 2023-09-15:
Exposed points in Niagara require a special permission to be written to. Something related to the user "bacnet" if I remember.

dqcontrols, 2023-09-15:
[image]
[image]
[image]

Here are the screenshots of the points setup, in descending order (Occupied Setpoint (point I want to write to) point discovery in Export Table, Occ. Setpoint Property Sheet in Export table, and the property sheet under Points). Seems like they can all be written to and have the permission from my understanding. Is the permission you are referring to at a higher level (i.e. at the Bacnet Network level)?
[image]

Thanks!

bbartling, 2023-09-16:
Your export table seems fine from what I can remember but its been a while for me using Workbench. Can you write to those points on the JACE with a BACnet scan tool okay? Verify that first, I use this all the time:
https://www.ccontrols.com/sd/bdt.htm

I always use BAC0 on a `bacnet = BAC0.lite()` type of app where then I write or read whole strings only, like this:
```
        write_vals = f'{address} {object_type} {object_instance} presentValue {temperature} - {priority}'
        bacnet.write(write_vals)
```

Am playing around with this at the moment of printing the weather web temperature value and writing to a BACnet device:
```
import requests
import BAC0
import time
import random

bacnet = BAC0.lite()

address = "192.168.0.101/24"
object_type = "analogValue"
object_instance = "1"
priority = 10  # BACnet priority

# Replace with the actual latitude and longitude of the desired location
location = '39.7456,-97.0892'

def get_current_temperature(location):
    points_url = f"https://api.weather.gov/points/{location}"
    response = requests.get(points_url)

    if response.status_code != 200:
        print("Failed to retrieve location data.")
        return None

    data = response.json()
    forecast_url = data["properties"]["forecast"]
    response = requests.get(forecast_url)

    if response.status_code != 200:
        print("Failed to retrieve forecast data.")
        return None

    data = response.json()
    temperature = data["properties"]["periods"][0]["temperature"]

    return temperature

while True:
    temperature = get_current_temperature(location)

    if temperature is not None:
        print(f"Current temperature for the location: {temperature}°F")

        # Write the retrieved temperature to the BACnet device
        write_vals = f'{address} {object_type} {object_instance} presentValue {temperature} - {priority}'
        bacnet.write(write_vals)
    else:
        print("Temperature data not available.")

    time.sleep(60)

bacnet.disconnect()

```

bbartling, 2023-09-18:
ah interesting Niagara `RoleService` ... nice to know if I run across that in the wild

dqcontrols, 2023-09-19:
Having some issues with writing to a specific priority. 
--------------------------------------------------------------------------------------------------------------
bacnet.read('10.0.0.92 analogValue 0 presentValue')
Out[7]: 75.0

bacnet.write('10.0.0.92 analogValue 0 presentValue 72 -8')
confirmed request encoding error: ValueError('value out of range')
Traceback (most recent call last):
  File "C:\Users\dqcontrols\anaconda3\lib\site-packages\bacpypes\appservice.py", line 1486, in sap_indication
    apdu.encode(xpdu)
  File "C:\Users\dqcontrols\anaconda3\lib\site-packages\bacpypes\apdu.py", line 695, in encode
    Sequence.encode(self, self._tag_list)
  File "C:\Users\dqcontrols\anaconda3\lib\site-packages\bacpypes\constructeddata.py", line 111, in encode
    helper = element.klass(value)
  File "C:\Users\dqcontrols\anaconda3\lib\site-packages\bacpypes\primitivedata.py", line 642, in __init__
    raise ValueError("value out of range")
ValueError: value out of range
Traceback (most recent call last):

  Cell In[8], line 1
    bacnet.write('10.0.0.92 analogValue 0 presentValue 72 -8')

  File ~\anaconda3\lib\site-packages\BAC0\core\io\Write.py:113 in write
    raise NoResponseFromController("APDU Abort Reason : {}".format(reason))

NoResponseFromController: APDU Abort Reason : Timeout

bacnet.write('10.0.0.92 analogValue 0 presentValue 72')

I am able read just fine and write without giving a specific priority, but attempting to write to ln8 fails.

bbartling, 2023-09-19:
i think you maybe missing a space in your string. Try changing:

`bacnet.write('10.0.0.92 analogValue 0 presentValue 72 -8')`

To this:

`bacnet.write('10.0.0.92 analogValue 0 presentValue 72 - 8')

maybe also verify in Niagara your point is writeable too?

closed 2023-10-04 by dqcontrols