ChristianTremblay/BAC0 #419: bad ReadRange request handling

MightyPork, 2023-09-19:
First of all, thanks for the library, its very handy for bacnet testing.

I found a bug in ReadRange handling, first it does not work with the "all" range (encoded by not passing any of the restriction parameters like index and count), and it also completely ignores the range parameters, so you always get the full list

https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/app/ScriptApplication.py#L203-L218

a simple fix to at least make "all" work:

```python
        if apdu.range is None:
            self._log.debug("    - read all")
        else:
            if apdu.range.byPosition:
                range_by_position = apdu.range.byPosition
                self._log.debug("    - range_by_position: %r", range_by_position)

            elif apdu.range.bySequenceNumber:
                range_by_sequence_number = apdu.range.bySequenceNumber
                self._log.debug(
                    "    - range_by_sequence_number: %r", range_by_sequence_number
                )

            elif apdu.range.byTime:
                range_by_time = apdu.range.byTime
                self._log.debug("    - range_by_time: %r", range_by_time)

            else:
                raise RejectException("missingRequiredParameter")
```

I also can't figure out how to define List properly so both ReadProperty and ReadRange work. It's my understanding of the spec that ReadProperty should work with a list, but I haven't tested this with a real bacnet device.

This works for ReadProperty:

```python
ObjectFactory(
    CalendarObject,
    10,
    "holidaysCalendar",
    properties={
        "presentValue": Boolean(False),
        "dateList": list([
            CalendarEntry(
                date=Date(month=12, day=24)
            ),
            CalendarEntry(
                date=Date(month=9, day=28)
            )
        ]),
    }
)
```

this does not work for ReadProperty, but works with ReadRange

```python
ObjectFactory(
    CalendarObject,
    10,
    "holidaysCalendar",
    properties={
        "presentValue": Boolean(False),
        "dateList": ListOf(CalendarEntry)(list([
            CalendarEntry(
                date=Date(month=12, day=24)
            ),
            CalendarEntry(
                date=Date(month=9, day=28)
            )
        ])),
    }
)
```

bbartling, 2023-09-19:
can you show how you run a ReadRange?

bbartling, 2023-09-19:
ill try to duplicate it on my end. I don't typically use ReadRange. What's the usecases on ReadRange?

MightyPork, 2023-09-19:
here's a packet dump from wireshark (remove .txt, renamed because github is stupid)
[readrange.pcapng.txt](https://github.com/ChristianTremblay/BAC0/files/12664870/readrange.pcapng.txt)

read range is used for list manipulation, together with add-list-element and remove-list-element, which are not implemented here