ChristianTremblay/BAC0 #421: How do I fetch data from a device which is inside a device?

Kabilan001, 2023-09-20:
I would like to start off with, big fan of your work.

Usually, in buildings when I do a whois and read the devices properties, I get all the objects belonging to that device. However, I have a new building where there is a child device within a parent device (according to yabe). When I do a whois and bacnet.properties(), I only get the properties of the parent device and not the child device. I have attached an image from yabe explaining the [architecture.](url) Have you experienced this situation before?
[image]

bbartling, 2023-09-20:
is the child device a twisted pair (MSTP) network downstream from the parent device?

if it is you maybe able to just read and write to the child devices directly. For example if the parent device is a BACnet router to downstream MSTP devices you should be able to just read the devices...if you can find out the network number of the child devices.

Just an assumption. If I run across a Siemens BAS HVAC controls it seems like they do this alot where an AHU is an IP based BACnet device where then all the terminal unit VAV boxes are a downstream twisted pair network but as long as the IP controller has routing enabled and can just to reads and writes.

thats just an assumption??? Else its something i never ran across before. Can you confirm its just one device?

bbartling, 2023-09-20:
ive never seen a device inside of a device but only down stream MSTP controllers

Kabilan001, 2023-09-20:
This is the first time I am seeing something like this as well. This is BACnet IP. So this is not like the Siemens one that you mentioned. The AHUs in my case are the child devices listed under the parent device. The child devices just have object name and instance ID.

bbartling, 2023-09-20:
whats the vendor of the equipment?

Kabilan001, 2023-09-20:
Automated logic is the BMS provider

bbartling, 2023-09-20:
seems very odd. I am also not familiar with Yabe.

What does it look like this tool? https://www.ccontrols.com/sd/bdt.htm

On a BACnet scan. Its free

Kabilan001, 2023-09-21:
When I do a 

bacnet.discover(networks='known')
print(bacnet.devices)

I get all the devices. When I do a whois, only the parent devices are getting listed.

bbartling, 2023-09-21:
can provide more code? And format it properly? For code snippets use the `brackets`. what tool are you using when it outputs devices in the data structure? `('device2451', 'Automated Logic Corporation', '24005:77@172.20.82.40', 2451)`?

what happens when you run this:

```
import BAC0, time

bacnet = BAC0.lite()
time.sleep(2)

devices = bacnet.whois(global_broadcast=True)
device_mapping = {}
addresses = []
for device in devices:
    if isinstance(device, tuple):
        device_mapping[device[1]] = device[0]
        print("Detected device %s with address %s" % (str(device[1]), str(device[0])))
print(device_mapping)
print((str(len(device_mapping)) + " devices discovered on network."))
```

Kabilan001, 2023-09-21:
This is my code which provides the output - ('device2451', 'Automated Logic Corporation', '24005:77@172.20.82.40', 2451)
`import BAC0
import bacpypes
import time
import sys

bacpypes.settings.settings['route_aware'] = True
#BAC0.log_level('silence')

bacnet = BAC0.connect("172.20.82.102/23")
#bacnet.whois()
bacnet.discover(networks='known')
print(bacnet.devices)`

This is the output when I run your code
`Detected device 2403 with address 172.20.82.30
Detected device 2523 with address 172.20.82.70
Detected device 2402 with address 172.20.82.20
Detected device 2404 with address 172.20.82.40
Detected device 2401 with address 172.20.82.10
Detected device 2517 with address 24004:71
Detected device 2440 with address 24003:9
Detected device 2518 with address 24004:72
Detected device 2436 with address 24003:2
Detected device 2437 with address 24003:3
Detected device 2400 with address 172.30.220.208
Detected device 2441 with address 24004:79
Detected device 2433 with address 24003:4
Detected device 2434 with address 24003:5
Detected device 2438 with address 24003:6
Detected device 2451 with address 24005:77
Detected device 2439 with address 24003:7
Detected device 2435 with address 24003:8
Detected device 2447 with address 24005:78
Detected device 2532 with address 24005:81
Detected device 2533 with address 24005:82
Detected device 2534 with address 24005:83
Detected device 2409 with address 24001:3
Detected device 2410 with address 24001:4
Detected device 2528 with address 24005:92
Detected device 2448 with address 24005:68
Detected device 2411 with address 24001:5
Detected device 2442 with address 24005:69
Detected device 2547 with address 24001:6
Detected device 2443 with address 24005:70
Detected device 2548 with address 24001:7
Detected device 2444 with address 24005:71
Detected device 2407 with address 24001:2
Detected device 2445 with address 24005:72
Detected device 2414 with address 24002:15
Detected device 2423 with address 24002:17
Detected device 2415 with address 24002:19
Detected device 2424 with address 24002:21
Detected device 2538 with address 24002:23
Detected device 2425 with address 24002:25
Detected device 2427 with address 24002:31
Detected device 2428 with address 24002:35
Detected device 2452 with address 24005:73
Detected device 2446 with address 24005:74
Detected device 2450 with address 24005:75
Detected device 2449 with address 24005:76
{2403: '172.20.82.30', 2523: '172.20.82.70', 2402: '172.20.82.20', 2404: '172.20.82.40', 2401: '172.20.82.10', 2517: '24004:71', 2440: '24003:9', 2518: '24004:72', 2436: '24003:2', 2437: '24003:3', 2400: '172.30.220.208', 2441: '24004:79', 2433: '24003:4', 2434: '24003:5', 2438: '24003:6', 2451: '24005:77', 2439: '24003:7', 2435: '24003:8', 2447: '24005:78', 2532: '24005:81', 2533: '24005:82', 2534: '24005:83', 2409: '24001:3', 2410: '24001:4', 2528: '24005:92', 2448: '24005:68', 2411: '24001:5', 2442: '24005:69', 2547: '24001:6', 2443: '24005:70', 2548: '24001:7', 2444: '24005:71', 2407: '24001:2', 2445: '24005:72', 2414: '24002:15', 2423: '24002:17', 2415: '24002:19', 2424: '24002:21', 2538: '24002:23', 2425: '24002:25', 2427: '24002:31', 2428: '24002:35', 2452: '24005:73', 2446: '24005:74', 2450: '24005:75', 2449: '24005:76'}
46 devices discovered on network.`

Kabilan001, 2023-09-21:
This is my code which provides the output - ('device2451', 'Automated Logic Corporation', '24005:77@172.20.82.40', 2451)
`import BAC0
import bacpypes
import time
import sys

bacpypes.settings.settings['route_aware'] = True
#BAC0.log_level('silence')

bacnet = BAC0.connect("172.20.82.102/23")
#bacnet.whois()
bacnet.discover(networks='known')
print(bacnet.devices)`

This is the output when I run your code
`
Detected device 2403 with address 172.20.82.30 Detected device 2523 with address 172.20.82.70 Detected device 2402 with address 172.20.82.20 Detected device 2404 with address 172.20.82.40 Detected device 2401 with address 172.20.82.10 Detected device 2517 with address 24004:71 Detected device 2440 with address 24003:9 Detected device 2518 with address 24004:72 Detected device 2436 with address 24003:2 Detected device 2437 with address 24003:3 Detected device 2400 with address 172.30.220.208 Detected device 2441 with address 24004:79 Detected device 2433 with address 24003:4 Detected device 2434 with address 24003:5 Detected device 2438 with address 24003:6 Detected device 2451 with address 24005:77 Detected device 2439 with address 24003:7 Detected device 2435 with address 24003:8 Detected device 2447 with address 24005:78 Detected device 2532 with address 24005:81 Detected device 2533 with address 24005:82 Detected device 2534 with address 24005:83 Detected device 2409 with address 24001:3 Detected device 2410 with address 24001:4 Detected device 2528 with address 24005:92 Detected device 2448 with address 24005:68 Detected device 2411 with address 24001:5 Detected device 2442 with address 24005:69 Detected device 2547 with address 24001:6 Detected device 2443 with address 24005:70 Detected device 2548 with address 24001:7 Detected device 2444 with address 24005:71 Detected device 2407 with address 24001:2 Detected device 2445 with address 24005:72 Detected device 2414 with address 24002:15 Detected device 2423 with address 24002:17 Detected device 2415 with address 24002:19 Detected device 2424 with address 24002:21 Detected device 2538 with address 24002:23 Detected device 2425 with address 24002:25 Detected device 2427 with address 24002:31 Detected device 2428 with address 24002:35 Detected device 2452 with address 24005:73 Detected device 2446 with address 24005:74 Detected device 2450 with address 24005:75 Detected device 2449 with address 24005:76 {2403: '172.20.82.30', 2523: '172.20.82.70', 2402: '172.20.82.20', 2404: '172.20.82.40', 2401: '172.20.82.10', 2517: '24004:71', 2440: '24003:9', 2518: '24004:72', 2436: '24003:2', 2437: '24003:3', 2400: '172.30.220.208', 2441: '24004:79', 2433: '24003:4', 2434: '24003:5', 2438: '24003:6', 2451: '24005:77', 2439: '24003:7', 2435: '24003:8', 2447: '24005:78', 2532: '24005:81', 2533: '24005:82', 2534: '24005:83', 2409: '24001:3', 2410: '24001:4', 2528: '24005:92', 2448: '24005:68', 2411: '24001:5', 2442: '24005:69', 2547: '24001:6', 2443: '24005:70', 2548: '24001:7', 2444: '24005:71', 2407: '24001:2', 2445: '24005:72', 2414: '24002:15', 2423: '24002:17', 2415: '24002:19', 2424: '24002:21', 2538: '24002:23', 2425: '24002:25', 2427: '24002:31', 2428: '24002:35', 2452: '24005:73', 2446: '24005:74', 2450: '24005:75', 2449: '24005:76'} 46 devices discovered on network.`

Kabilan001, 2023-09-21:
This is my code which provides the output - ('device2451', 'Automated Logic Corporation', '24005:77@172.20.82.40', 2451)

```
import BAC0
import bacpypes
import time
import sys

bacpypes.settings.settings['route_aware'] = True
#BAC0.log_level('silence')

bacnet = BAC0.connect("172.20.82.102/23")
#bacnet.whois()
bacnet.discover(networks='known')
print(bacnet.devices)`
```

This is the output when I run your code
`Detected device 2403 with address 172.20.82.30 Detected device 2523 with address 172.20.82.70 Detected device 2402 with address 172.20.82.20 Detected device 2404 with address 172.20.82.40 Detected device 2401 with address 172.20.82.10 Detected device 2517 with address 24004:71 Detected device 2440 with address 24003:9 Detected device 2518 with address 24004:72 Detected device 2436 with address 24003:2 Detected device 2437 with address 24003:3 Detected device 2400 with address 172.30.220.208 Detected device 2441 with address 24004:79 Detected device 2433 with address 24003:4 Detected device 2434 with address 24003:5 Detected device 2438 with address 24003:6 Detected device 2451 with address 24005:77 Detected device 2439 with address 24003:7 Detected device 2435 with address 24003:8 Detected device 2447 with address 24005:78 Detected device 2532 with address 24005:81 Detected device 2533 with address 24005:82 Detected device 2534 with address 24005:83 Detected device 2409 with address 24001:3 Detected device 2410 with address 24001:4 Detected device 2528 with address 24005:92 Detected device 2448 with address 24005:68 Detected device 2411 with address 24001:5 Detected device 2442 with address 24005:69 Detected device 2547 with address 24001:6 Detected device 2443 with address 24005:70 Detected device 2548 with address 24001:7 Detected device 2444 with address 24005:71 Detected device 2407 with address 24001:2 Detected device 2445 with address 24005:72 Detected device 2414 with address 24002:15 Detected device 2423 with address 24002:17 Detected device 2415 with address 24002:19 Detected device 2424 with address 24002:21 Detected device 2538 with address 24002:23 Detected device 2425 with address 24002:25 Detected device 2427 with address 24002:31 Detected device 2428 with address 24002:35 Detected device 2452 with address 24005:73 Detected device 2446 with address 24005:74 Detected device 2450 with address 24005:75 Detected device 2449 with address 24005:76 {2403: '172.20.82.30', 2523: '172.20.82.70', 2402: '172.20.82.20', 2404: '172.20.82.40', 2401: '172.20.82.10', 2517: '24004:71', 2440: '24003:9', 2518: '24004:72', 2436: '24003:2', 2437: '24003:3', 2400: '172.30.220.208', 2441: '24004:79', 2433: '24003:4', 2434: '24003:5', 2438: '24003:6', 2451: '24005:77', 2439: '24003:7', 2435: '24003:8', 2447: '24005:78', 2532: '24005:81', 2533: '24005:82', 2534: '24005:83', 2409: '24001:3', 2410: '24001:4', 2528: '24005:92', 2448: '24005:68', 2411: '24001:5', 2442: '24005:69', 2547: '24001:6', 2443: '24005:70', 2548: '24001:7', 2444: '24005:71', 2407: '24001:2', 2445: '24005:72', 2414: '24002:15', 2423: '24002:17', 2415: '24002:19', 2424: '24002:21', 2538: '24002:23', 2425: '24002:25', 2427: '24002:31', 2428: '24002:35', 2452: '24005:73', 2446: '24005:74', 2450: '24005:75', 2449: '24005:76'} 46 devices discovered on network.`

bbartling, 2023-09-22:
a device like `'24005:73'` is a twisted pair network (RS-485 - old stuff) MSTP device.

Am not sure what Yabe is trying to show in how some devices are underneath another....unless its a Tree structure or something but I would just treat all the devices as individual unique devices...the `46 devices discovered on network.`

I think that network is real similar to what I work all the time, see these examples for how I use BAC0 in simple stuff of reads and writes to present value of HVAC controls I/O and setpoints.

https://gist.github.com/bbartling/88e95557b0b297bea721c0f285657d53

Kabilan001, 2023-09-25:
Yes this works. Thank you so much for your time.

closed 2023-09-25 by Kabilan001