ChristianTremblay/BAC0 #423: Accessing 'Network_Port' and 'Loop' objects

dqcontrols, 2023-10-04:
Hi,

I am encountering more devices that have Network_Port and Loop objects. How can I access (i.e. read/write) to these points using BAC0?

bbartling, 2023-10-04:
Am sort of guessing here but take a peak at bacpypes repo objects.py: (maybe try a Find and search)
https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py

Which I think BAC0 accesses under the hood...make sure your objects are at least in object.py and then paste your code in a `formatted way` that other people can glance at.

ChristianTremblay, 2023-10-05:
by using the bacpypes syntax `bacnet.read("your stuff here")`you can probably end up with soemthing.
I never work with Loop object in my day job...so I never put efforts on that.

Time being a super rare resources those days, I cannot spend a lot of time developing something new... but PR are welcomed...even if I can be very slow sometimes.

dqcontrols, 2023-10-10:
> Am sort of guessing here but take a peak at bacpypes repo objects.py: (maybe try a Find and search) https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py
>
> Which I think BAC0 [...]

Yes, this pretty much works for me. Thanks!