ChristianTremblay/BAC0 #424: Error when instanciating TrendLog

Rodrigo-Ornellas, 2023-10-12:
Hello all,

As many have mentioned, great work with BAC0.
I am still begging to use the library but I am open to help out with supporting it (in case help is needed).

I am trying to read a TrendLog from a Controller using the instructions from BAC0 documentation.
https://bac0.readthedocs.io/en/latest/histories.html#trendlog

The error is happening when I instanciate a TrendLog. 
Here is what I am getting from the execution:

```shell
2023-10-11 16:51:22,752 - INFO    | Starting BAC0 version 23.07.03 (Lite)
2023-10-11 16:51:22,752 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2023-10-11 16:51:22,752 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2023-10-11 16:51:22,752 - INFO    | Starting TaskManager
2023-10-11 16:51:22,754 - INFO    | Using ip : 172.16.3.60 on port 47808 | broadcast : 172.16.3.255
2023-10-11 16:51:22,799 - INFO    | Starting app...
2023-10-11 16:51:22,800 - INFO    | BAC0 started
2023-10-11 16:51:22,801 - INFO    | Registered as Simple BACnet/IP App
2023-10-11 16:51:22,801 - INFO    | Device instance (id) : 3056951
2023-10-11 16:51:22,807 - INFO    | Update Local COV Task started (required to support COV)
2023-10-11 16:51:22,808 - INFO    | Changing device state to DeviceDisconnected'>
2023-10-11 16:51:22,818 - INFO    | 172.16.3.109 router to [4]
2023-10-11 16:51:23,009 - INFO    | Changing device state to RPMDeviceConnected'>
2023-10-11 16:51:23,341 - INFO    | Device 21122:[TEC3000-99] found... building points list
2023-10-11 16:51:38,158 - WARNING | APDU Abort Reject Reason : Timeout
2023-10-11 16:51:38,159 - WARNING | No response from controller Timeout
2023-10-11 16:51:44,541 - ERROR   | Device creation failed... re-connecting
2023-10-11 16:51:44,542 - INFO    | Changing device state to DeviceDisconnected'>
2023-10-11 16:51:44,911 - INFO    | Changing device state to RPMDeviceConnected'>
2023-10-11 16:51:45,502 - INFO    | Device 21122:[TEC3000-99] found... building points list
2023-10-11 16:52:02,688 - INFO    | Ready!
2023-10-11 16:52:02,689 - INFO    | Device defined for normal polling with a delay of 10sec
2023-10-11 16:52:02,689 - INFO    | Polling started, values read every 10 seconds
Traceback (most recent call last):
  File "//test.py", line 28, in <module>
    trend = BAC0.TrendLog(instance, device)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/BAC0/core/devices/Trends.py", line 80, in __init__
    self.read_log_buffer()
  File "/usr/local/lib/python3.11/site-packages/BAC0/core/devices/Trends.py", line 125, in read_log_buffer
    start = max(_actual_index - self.properties.record_count, self._last_index)
                                                              ^^^^^^^^^^^^^^^^
AttributeError: 'TrendLog' object has no attribute '_last_index'
``` 

When I looked into the code from Trends.py (indicated in the error), it looked to me like the self.read_log_buffer method tries to access the self._last_index property before it has been defined in the __init__ function.

I tested moving the self._last_index 2 lines up before the IF statement (inside the __init__ function) and it seemed to have resolved the problem but I only ran a few tests.

Here is the code that I used:

```shell
import BAC0

bacnet = BAC0.lite(ip='192.168.10.11/24')
address = '4:99'
device_id =  21122

# Intanciate Device/Controller
device = BAC0.device(address,device_id,bacnet)

# Instanciate the TrendLog
instance = 101010
trend = BAC0.TrendLog(instance, device)
``` 

Has anyone encountered similar problem?
Is this really an error or am I using the TrendLog incorrectly?
Can anyone help me?

Thanks
Rod

bbartling, 2023-10-12:
can you verify with a 3rd party tool that the `device` has a `TrendLog` object? I personally have never worked with a BACnet trendlog object...

I use this free tool all the time if I need to troubleshoot something that wont work with a script: https://www.ccontrols.com/sd/bdt.htm

Or YABE is another option

Rodrigo-Ornellas, 2023-10-12:
Hi Ben,
Thanks for helping.
I have a couple of BACnet devices that I have tested with and I know that they have TrendLog objects.
And as I mentioned in my original message, when I make the modification in the BAC0 code, I am able to read the TrendLog data.

This is the BAC0 file that I modified:
```shell
/usr/local/lib/python3.11/site-packages/BAC0/core/devices/Trends.py
``` 

And here is the Original Code that does NOT seem to be working (at least for me):
```shell
@note_and_log
class TrendLog(TrendLogProperties):
    """
    BAC0 simplification of TrendLog Object
    """

    def __init__(
        self, OID, device=None, read_log_on_creation=True, multiple_request=None
    ):
        self.properties = TrendLogProperties()
        self.properties.device = device
        self.properties.oid = OID
        self.update_properties()

        if read_log_on_creation:
            self.read_log_buffer()
        self._last_index = 0
``` 

And here is the code AFTER I changed it and it started to work:
```shell
@note_and_log
class TrendLog(TrendLogProperties):
    """
    BAC0 simplification of TrendLog Object
    """

    def __init__(
        self, OID, device=None, read_log_on_creation=True, multiple_request=None
    ):
        self.properties = TrendLogProperties()
        self.properties.device = device
        self.properties.oid = OID
        self.update_properties()
        self._last_index = 0            # <<<<<< I moved it here
        if read_log_on_creation:
            self.read_log_buffer()
        #self._last_index = 0             <<<<< I left this here just to illustrate where it was before

``` 

Sorry, I was probably not very clear in my original message.
Thanks again for the help.

bbartling, 2023-10-12:
nice work. What does trend log object do in your application?

I have only read data with read requests and then saved the data to my own data base.. but does a trend log object save the data and then in BACnet you can just request that data?

Rodrigo-Ornellas, 2023-10-12:
I am still learning about BACnet in general. I hope that my explanation is correct in its content.

As far as I understand, a TrendLog works similar to  a buffer of data from one specific point of the controller over a longer period of time. I have heard people referring to a TrendLog as a Historian because it keeps the history of values of that point. 

The buffer is not large and it depends on what is the interval into which the data is set to be collected. Shorter intervals (every minute for example) will generate a lot of samples and therefore generate a shorter period of time that the buffer can hold that data. If you use longer intervals like every 5 or 15 minutes, then the buffer (TrendLog) will be able to store data for longer period of time. This is a circular buffer as far as I understand. So the data will be overwritten after sometime.

My application is interested in collecting these longer period data to understand what changes have happened to these points.
I can collect data from individual points at the 15 minutes interval and build my own TrendLog or I can try to collect the TrendLog that was already built by the Controller for me (less work).

Example: AnalogInput is an object that I can read a sample. But I can check the controller and see if there is a TrendLog that has been set to collect data for this AnalogInput.

I hope my understanding is correct.
Although I was able to collect the data (after the fix that I mentioned above) from the TrendLog, I am now struggling to identify which POINT this TrendLog is collecting data from (This would be my next question here).

bbartling, 2023-10-12:
That's a great question... I wouldn't be surprised if that is something setup by the person whom installed the BACnet device.

When I worked for JCI in the field using proprietary JCI software ... I dont remember being able todo a BACnet trend log object in the software I used to setup a JCI HVAC system... but it wouldnt surprise me if a person can setup a trend log object in the device they are deploying where then someone like you on a BACnet scan of the system, how do you figure what data is being logged inside of that trend log especially if the BACnet device has many different sensors or processes going on in it.

Rodrigo-Ornellas, 2023-10-12:
I was told that there is a property in the TrendLog Object that references to which Object (other object) this TrendLog is trending for. I was not told which property that is but in the BACnet specification (ASHRAE) I found a TrendLog property ( "Log_DeviceObjectProperty" page 281) that I think indicates the Original Object that is being recorded by the TrendLog.

I am trying to find a way to read this property now.

bbartling, 2023-10-12:
TrendLog is a huge mystery to me on use case other than maybe it could be
used to cut down on bacnet traffic..?. For example if the device samples
data on very short intervals internally where then a trend log point could
be an average of that data, then it makes sense where then IoT wouldn't
need to read request on some short intervals to get the same results.

Is that what TrendLog is used for? I'll ask this to ace IoT discord channel.

Rodrigo-Ornellas, 2023-10-12:
Thanks for sharing the link to the discord channel. That will be very helpful.

In my application, we analyse the trend over time for a range of selected points to try to generate some intelligence to the building by interpreting long term data.

Reading TrendLogs is more cost effective in terms of the hardware, time to read, traffic and etc. It is more efficient in my case than reading the same point many times a day.

I was able to read the "Log_DeviceObjectProperty" and identify to which point is the TrendLog associated.

I am still concerned with the issue which started this thread.
I looking for some help to clarify if this is an issue with the BAC0 library or is it me doing something wrong.

Appreciate your engagement!
Really cool. Thanks

ChristianTremblay, 2023-10-13:
Please pull the develop version. I fixed the code including your proposal and I corrected the sequence number that was Off...

Once done, for a JCI TEC, I suggest you try this : 

```python
import BAC0
bacnet = BAC0.lite(ip='192.168.211.16/24') # your IP
from BAC0.core.proprietary_objects.jci import tec_short_point_list
obj_lst = tec_short_point_list() # this will speed up the creation by using only typical variables from the TEC
tec = BAC0.device('303:7', 5007, bacnet, object_list=obj_lst)
tec['ZNT_Trend'].history
```

`ZNT_Trend` is a BACnet TrendLog object. Which is an object that store a timeseries of the zone temperature. No need to create the object as BAC0.device will create it for you.

To kow what is the object being recorded by the trendLog, you can use the `logDeviceObjectProperty'. But his will not always give you something useful. For example, in the TEC...

In [13]: bacnet.read('303:7 trendLog 101010 logDeviceObjectProperty').debug_contents()
    objectIdentifier = (792, 13)
    propertyIdentifier = 7000

This point to a proprietary object, hidden inside the controller.

closed 2023-12-20 by github-actions[bot]