ChristianTremblay/BAC0 #427: BAC0 complete is not starting, only lite works always

Piruthivi, 2023-10-20:
_I am new to python and BAC0, am trying to use this module for my device.
i made a simple python script to communicate with my device._

_import time
import BAC0
a = bacnet.write("192.168.2.100 binaryValue 1 presentValue active")
print("write AV1 binaryValue active")
time.sleep(1)
a = bacnet.write("192.168.2.100 analogValue 1 presentValue 20")
print("write AV1 analogValue 10")
time.sleep(1)
a = bacnet.read("192.168.2.100 analogValue 1 presentValue")
print("write AV1 analogValue ",a)
time.sleep(1)
a = bacnet.subscribe_cov("192.168.2.100",("analogInput", 1))
print(a)_

**and am getting the same output in both windows 11 and ubuntu 22 version, which is** 

_================================================================================
Import Error, refer to documentation or reinstall using
    $ "pip install BAC0"
 cannot import name 'Figure' from 'bokeh.plotting' (C:\Users\u422739\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bokeh\plotting\__init__.py)

_Discard this message if you are actually installing BAC0._
================================================================================
_Traceback (most recent call last):
  File "D:\Python_workout\motor_current_test.py", line 4, in <module>
    bacnet = BAC0.connect(ip='192.168.2.60/24')
AttributeError: module 'BAC0' has no attribute 'connect'_

to run complete mode i already installed flask, flask-bootstrap, bokeh and pandas using pip install
Please help me how to solve this issue. so that i could move forward

ChristianTremblay, 2023-12-11:
You probably don't need to "complete" and please note that there are bugs related to newer version of Bokeh... and I'm planning to deprecate this part of the software. Using Grafana and InfluxDB for example is way easier and more robust than what I did.

magicmatt007, 2024-01-28:
I installed BAC0 in a virtual environment and used `pip install BAC0`

and then installed the other recommended packages also with  `pip flask, flask-bootstrap,bokeh,pandas`

This installed the latest bokeh version 3.3.4 and threw above error. Forcing PIP to use an older version of bokeh (I randomly selected the last 2.x version 2.4.3) fixed the problem:

```
pip uninstall bokeh
pip install bokeh==2.4.3.
```

So maybe give a hint at BAC0, which versions of the additional packages should be used....

ChristianTremblay, 2024-02-29:
Next version of BAC0 drops bokeh and everything else related to the web server.

closed 2024-05-06 by github-actions[bot]