ChristianTremblay/BAC0 #430: AttributeError: module 'BAC0' has no attribute 'lite

mbulinski, 2023-11-07:
Hello
a have a problem when try to run a project read-data.py

requirements are instaled:
BAC0==21.12.3
bacpypes==0.18.6
colorama==0.4.4

bbartling, 2023-11-07:
what does your script look like? `bacnet = BAC0.lite()` usually works for me

mbulinski, 2023-11-07:
thank You for your answer

Standard downloaded from github

import BAC0
import json
import time

# create a bacnet connection to connect to the devices,
# use your local machine IP address and the subnet below and also specify the port number if needed
bacnet = BAC0.lite(ip="192.168.18.6/24", port=47809)
data = {}

def main():
    while True:
        # Reading 3 Objects from the Bacnet Simulator
        for x in range(0, 3):
            id = str(x)
            # Connect to the BACnet simulator using the IP address and port number
            value = bacnet.read(
                "192.168.18.10:53522 analogInput " + id + " presentValue")
            value = str(value)
            data["Analog_input" + id] = value

        print(json.dumps(data, indent=2))
        time.sleep(5)

# RUN main
if __name__ == '__main__':
    main()

bbartling, 2023-11-07:
you on windows or linux? if you are on Linux just use `python` else just `py` for Windows.

Its to see if you can import it okay

```
> py
Python 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import BAC0
>>> bacnet = BAC0.lite()
```
see if that works

mbulinski, 2023-11-07:
Windows 10

Python 3.12.0 (tags/v3.12.0:0fb18b0, Oct  2 2023, 13:03:39) [MSC v.1935 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import BAC0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'BAC0'
>>> bacnet = BAC0.lite()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'BAC0' is not defined
>>>

bbartling, 2023-11-07:
try a `py -m pip intall BAC0` and see if it installs okay. then try again, your script may work as well

bbartling, 2023-11-07:
on windows if you have multiple version of python you can call it like `py -3.10` or `py 3.11` or `py -3.12` or whatever...

and you can run your script like `py -3.12 your_script.py` if need to specify a python version

mbulinski, 2023-11-07:
> try a `py -m pip intall BAC0` and see if it installs okay. then try again, your script may work as well

[image]

mbulinski, 2023-11-07:
> on windows if you have multiple version of python you can call it like `py -3.10` or `py 3.11` or `py -3.12` or whatever...
> 
> and you can run your script like `py -3.12 your_script.py` if need to specify a python version

[image]

bbartling, 2023-11-07:
actually you know what the issue is its python 3.12

there is an overhaul of python on version 3.12 they got rid of the asnycore module so @ChristianTremblay is rewriting BAC0 with bacpypes3 which is asyncio and not asyncore.

Try this install python 3.11:
https://www.python.org/downloads/release/python-3116/

I just download and run the .exe file version and add it to PATH when prompted:
[image]

Do all the processes again with the pip installl run it all on `py -3.11` and it should work

mbulinski, 2023-11-07:
It's getting better :)
[image]

bbartling, 2023-11-08:
that message says there is another software program on your computer running on port 47808. I get this if i am running a BACnet scan tool and then try to run a BAC0 script at the same time... the operating system says no you cannot do that.

I always running BACnet applications on completely different machines. If you are running multiple BACnet applications on the same machine and using like a BBMD or something to route the data.. ive never done that before its over my head.

what is the topology you are attempting?

closed 2024-01-15 by github-actions[bot]