ChristianTremblay/BAC0 #433: How to subscribe COVProperty using BAC0?

Chris-Wang-bjc, 2023-12-04:
Hi, I want to use BAC0 to implement the service about subscribe COVProperty, but I only found the function to subscribe COV, so I want to know how to subscribe COVProperty with BAC0?

Chris-Wang-bjc, 2023-12-05:
I'm glad to receive your reply. Yes, I have reviewed this document, but I have not found any information about subscribe COVProperty from it. I mean, how do I use BAC0 to subscribe COVProperty? I only found the function for subscribe COV, not the function for subscribe COVProperty.

bbartling, 2023-12-05:
chat GPT to the rescue ... which i way over use.

```
import BAC0

# Connect to the BACnet network
bacnet = BAC0.lite()

# Define the device and property you want to monitor
device_address = '192.168.1.101'  # Replace with the actual address of your BACnet device
object_type = 'analogInput'       # The type of BACnet object (e.g., 'analogInput', 'binaryOutput', etc.)
instance_number = 1               # The instance number of the object
property_name = 'presentValue'    # The property you want to monitor (e.g., 'presentValue')

# Create a COV subscription
cov_subscription = bacnet.cov(
    device_address,
    object_type,
    instance_number,
    property_name,
    duration=60  # Duration in seconds, 0 for indefinite
)

# Handle updates
def on_cov_update(data):
    print('COV Update:', data)

# Subscribe to COV updates
cov_subscription.subscribe(on_cov_update)

# Keep the script running to monitor updates
try:
    while True:
        pass
except KeyboardInterrupt:
    # Unsubscribe and clean up on exit
    cov_subscription.unsubscribe()
    bacnet.disconnect()
```

ill try tinkering with this on my test bench as well

Chris-Wang-bjc, 2023-12-07:
Thank you very much. I'll try.

bbartling, 2023-12-07:
I didn't have any luck post something if you have success

ChristianTremblay, 2023-12-11:
This is not implemented. cov.py would need to be modified to include a SubscribeCOVPropertyRequest

closed 2024-02-17 by github-actions[bot]