ChristianTremblay/BAC0 #438: How do I get a UnconfirmedCOVNotification without subscription

duxincai, 2024-01-09:
My bacnet server does not support subcription , but broadcast an UnconfirmedCOVNotification when cov occured.
I see subscriberProcessIdentifier is necessary In [do_UnconfirmedCOVNotificationRequest](https://github.com/ChristianTremblay/BAC0/blob/a6e3d3dfa18d96eb3e567196e15bbcfef82d9065/BAC0/core/app/ScriptApplication.py#L151C32-L151C32) .

How do I resolve it in this case?

bbartling, 2024-01-09:
what are you trying to resolve?

ChristianTremblay, 2024-01-18:
I have no idea... maybe you could find something about that on bacpypes directly

duxincai, 2024-01-23:
I want to recieve the broadcast UnconfirmedCOVNotification and show the value in web pages.
Now, I add a subscription_broadcast_cov in cov.py, and call it in my app after BAC0.lite().
It's worked.

```python
self._bacnet_instance = BAC0.lite(ip=connectionInfo["myIp"], localObjName=self.module_ID)
self._bacnet_instance.subscription_broadcast_cov(callback=self.cov_handler)
```

```python
    def subscription_broadcast_cov(self, callback):
        context = SubscriptionContext(
            address="*",
            objectID=None,
            confirmed=False,
            callback=callback,
        )
        self.subscription_contexts["broadcast_callback_context"] = context
        if "context_callback" not in self.subscription_contexts.keys():
            self.subscription_contexts["context_callback"] = self.context_callback
```

The subscription_contexts["broadcast_callback_context"] is using in ScriptApplication.py#do_UnconfirmedCOVNotificationRequest

```python
    def do_UnconfirmedCOVNotificationRequest(self, apdu):
        # look up the process identifier
        context = self.subscription_contexts.get(apdu.subscriberProcessIdentifier, None)

        if (not context) and (apdu.subscriberProcessIdentifier == 0 or apdu.pduDestination == "*") and "broadcast_callback_context" in self.subscription_contexts.keys():
            context = self.subscription_contexts["broadcast_callback_context"]

        if not context or (context.address != "*" and apdu.pduSource != context.address):
            return
...
```

closed 2024-01-23 by duxincai