ChristianTremblay/BAC0 #439: Write on local device with external software

Sekiro-kost, 2024-01-15:
Hello

I have created a script, to create a BACnet device, to add values coming from different sources (loraWan, zigbee, etcc) on this bacnet device, on my Linux machine.
I can also update these bacnet points via :

`my_device["Current_TempLora"].presentValue = 24`

No problem, I can read this information via a third-party program on another machine, on the same network.

However, I can't write to these bacnet points from an external program.

Here's how my device and its points are created:

```
def start_device(device, arrayCmds):
    logging.debug("Starting BACnet device")
    _new_objects = []
    try:
        for x in arrayCmds:
            formatVariableEnter = type(x['PresentValue'])
            if formatVariableEnter is str:
                continue

            new_object = analog_input(  
                instance=int(x['CmdId']),
                name=str(x['NameCmd']),
                description=str(x['Description']),
                presentValue=formatVariableEnter(x['PresentValue']),
                properties={"units": str(x['Unite'])},
            )
            
            _new_objects.append(new_object)

        for obj in _new_objects:
            obj.add_objects_to_application(device)

        return device
    except Exception as e:
          logging.error('ERREUR DANS CONSTRUCTION DEVICE BACNET AVEC LA COMMANDE >>>' + str(x['NameCmd']) + '. VERIFIEZ SES PARAMETRES (UNITES etc..) : ' + str(e))
```

Sekiro-kost, 2024-01-16:
I also tried this :

```
def start_device(device, arrayCmds):
    logging.debug("Starting BACnet device")
    av = []
    try:
        for x in arrayCmds:
            logging.debug(x)

            new_object = create_AV(oid=int(x['CmdId']), name=str(x['NameCmd']), pv=0, pv_writable=True)
            new_object.units = EngineeringUnits(str(x['Unite']))
            new_object.description = CharacterString(str(x['Description']))
            new_object.presentValue = float(x['PresentValue'])         
            av.append(new_object)

        for each in av:
            device.this_application.add_object(each)

        logging.debug(av)        
        return device
```

With my other software, I try to write on this point, but nothing happens.
It works only in my script, I attempt to write with av.presentValue = x.x

bbartling, 2024-01-16:
Maybe try glancing at this older issue for a fake air handling unit:
https://github.com/ChristianTremblay/BAC0/issues/381

I did all my points with a flag True for `is_commandable`:
```
# create AHU duct pressure setpoint point
_new_objects = analog_value(
    name="DAP-SP",
    properties={"units": "inchesOfWater"},
    description="AHU Duct Pressure Setpoint",
    presentValue=1, is_commandable=True
)
```

Sekiro-kost, 2024-01-17:
> Maybe try glancing at this older issue for a fake air handling unit: #381
>
> I did all my points with a flag True for `is_commandable`:
>
> ```
> # create AHU duct pressure setpoint point
> _new_objects = [...]

Thank you, I will test your solution

Sekiro-kost, 2024-01-17:
In my logs, I can see that when I tried to write a null value from my external software : 

`[2024-01-17 11:01:12]DEBUG :     - execution reject: InvalidParameterDatatype('presentValue must be of type Real')`

But If I try a valid value, I no longer get the error, but the value doesn't change on my bacnet device, even with your suggested method

ChristianTremblay, 2024-01-18:
Juste être sûr de comprendre....
You want to create an analog value object in your device on which you could write, via BACnet, from another device. Right ? Like a setpoint.

closed 2024-03-27 by github-actions[bot]