ChristianTremblay/BAC0 #440: BAC0 with Docker

Rahul14189, 2024-01-23:
Hi,
Thanks for the library!
i am trying to use this with docker. If i use this library on local machine directly, it works well, but if i try to use Docker, then it starts only with the ipaddress of the docker, and does not discover devices on local wifi network or local machine and can't read as well. If i try to give the local machine ip-address then it throws error as,
[image]

i have tried different settings like, network_mode: host, network_mode: bridge, etc.   but it did not work.
i tried to use bacnet = BAC0.connect(ip='xxx.xxx.xxx.xxx/mask', bbdmAddress=bbmdIP, bbmdTTL=bbmdTTL), it also did not work.
Can you please guide, what am i missing here !!

bbartling, 2024-01-23:
bacnet with Docker is really tough... what happens if you just pass in `bacnet = BAC0.lite()` without passing in any args?

Ive experimented a tiny bit with bacpypes3 and something like this appeared to work for an experimental BACnet server app:
```
# Use an Alpine Linux base image for Raspberry Pi
FROM arm32v7/alpine:latest

# Set the working directory in the container
WORKDIR /app

# Install necessary packages and dependencies
RUN apk --no-cache add python3 python3-dev py3-pip

# Install BACpypes and any other required dependencies
RUN pip3 install bacpypes3

# Copy your Python application into the container
COPY mini_device_revisited.py /app/mini_device_revisited.py

# Expose the UDP port 47808
EXPOSE 47808/udp

# Define the command to run your BACpypes application
# CMD ["python3", "/app/mini_device_revisited.py", "--debug", "bacpypes3.ipv4.IPv4DatagramProtocol"]
CMD ["python3", "/app/mini_device_revisited.py", "--debug"]
```

Rahul14189, 2024-02-02:
> bacnet with Docker is really tough... what happens if you just pass in `bacnet = BAC0.lite()` without passing in any args?
>
> Ive experimented a tiny bit with bacpypes3 and something like this [...]

Thanks for the reply,  it worked with the above code, i had to change the docker image to python 3.9. I did a bit of experiment with the code, and found that it does not discover the devices on the network but if we provide it the address of the device with tag detail then it is able to read. So, we can perform read and bulk read using docker. As docker and machine both gets different ip which creates the constraint BAC0 to auto detect the devices and there tags over the wifi network. i am looking for a way to overcome this, if i get a way i will update the progress.

Thanks @bbartling

ChristianTremblay, 2024-02-02:
The issue with discovery is due to the UDP broadcast that can't travel between the docker instance and the other device. 

Direct IP unicast messages will work. 

But not request like Whois, iam, who has , whoisroutertonetwork...

If you know what to point (a specific controller IP) you are good. 

It is also true for any device that would be exposed on the internet. You will not be able to discover them... but if you know their IP it works. This tells you why you should not expose bacnet ip controller without any security.

closed 2024-02-06 by Rahul14189