ChristianTremblay/BAC0 #442: Simulated device not sending COV

Weird-Banana, 2024-02-04:
Hi, I am using BAC0 22.9.21 to simulate a BACnet device with different AnalogInput points.
In the same script I have a function that randomly changes the present value of these points, till here everything works fine.

I have now created another script with BAC0 which reads the simulated device and use bacnet.cov(...) to subscribe to my simulated points. However, I am not receiving notifications on when the present value changes.

Is there anything I need to add to my simulator to make my points subscribable?

Weird-Banana, 2024-02-04:
I have removed the lifetime parameter from bacnet.cov and now it seems like that the callback is always getting triggered although there was no COV.

ChristianTremblay, 2024-02-07:
The local object in a BAC0 instance do not support COV yet. The work done covers receiving COV... not sending.
I think it'll be possible with the next iteration using bacpypes3.

Weird-Banana, 2024-02-13:
Thanks for the clarification, I thought I was doing something with the simulation part :)

closed 2024-02-13 by Weird-Banana