ChristianTremblay/BAC0 #444: In BAC0, Is it possible to expose actual points to Bacnet Server Points ?

Daniel-V-Richardson, 2024-02-19:
Hello,
I don't know whether my question makes any sense, but i have a requirement and i don't know how to achieve it. 
 
I have Created a network with the below details :
`client = BAC0.lite(ip=networkAddress, port=networkPort, deviceId=device_id)`
After this i am discovering devices and adding them to my db and then adding points for each device respectively to my db aswell. I am using Room Control Simulator as a Bacnet Device and it's points. 

Now I need to export these Imported DB Points as Bacnet points based on the client which i had created before, and i can do this by using object factory, which is working fine. But my requirement is, I need to establish a link between the exported bacnet point and the actual point. Which means if i try to update the setpoint1 of the exported point. then it should also reflect on the actual setpoint1.

I am trying to achieve this functionality for a long time but I am a beginner to field and i don't really get the entire picture. Please Help me out! 

Thanks.

bbartling, 2024-02-19:
maybe check out #381 for the fake AHU simulator code. Run it on some device and then on another device on your LAN discover it and the points. Don't test 2 BBACnet applications on the same OS but 2 separate devices/hardware networked together.

https://github.com/ChristianTremblay/BAC0/issues/381

hr-shloklabs, 2024-03-30:
@bbartling This script is just trying to simulate a bacnet device which is then discovered by another device. But want i wanted was kinda different. For example I have a bacnet device let's say an AHU device exposed on port 47808 simulated by niagra, and i have discovered it and enabled polling in my BAC0 application which used BAC0 library, now i want to export the same points of the simulated AHU on port 47809 which will be exposed as BAC1 points from my application. Now if i dicover the BAC1 device's points in another device i want the actual polling of the Original AHU's points to be happening, and also if i try to write to any point it should write not only the exported point from BAC0 device but the original point too. I don't know if i'm clear enough but this is what my understanding and requirement is. Thank You for your reply.

Daniel-V-Richardson, 2024-03-30:
@bbartling This script is just trying to simulate a bacnet device which is then discovered by another device. But want i wanted was kinda different. For example I have a bacnet device let's say an AHU device exposed on port 47808 simulated by niagra, and i have discovered it and enabled polling in my BAC0 application which used BAC0 library, now i want to export the same points of the simulated AHU on port 47809 which will be exposed as BAC1 points from my application. Now if i dicover the BAC1 device's points in another device i want the actual polling of the Original AHU's points to be happening, and also if i try to write to any point it should write not only the exported point from BAC0 device but the original point too. I don't know if i'm clear enough but this is what my understanding and requirement is. Thank You for your reply.

bbartling, 2024-03-30:
I think you need a BBMD, right @ChristianTremblay/BAC0
***@***.***> ?

You can run 2 BACnet apps on one OS but I they need a BBMD to route data
between them. I have actually never done this before.

That's what you want right? Run 2 apps on one machine and then read/write
between the two?

ChristianTremblay, 2024-03-30:
Not sure why you need that. From BAC1, you can point directly to BAC0 (the port, not the app) using unicast (a direct connection). I see no need for an intermediate device.

BBMD are used to make broadcast messages work on different subnets. If you know where to connect and don't need broadcast, a direct connection will work.

Daniel-V-Richardson, 2024-03-31:
> I think you need a BBMD, right @ChristianTremblay/BAC0 ***@***.***> ? You can run 2 BACnet apps on one OS but I they need a BBMD to route data between them. I have actually never done this before. That's what you want right? Run 2 apps on one machine and then read/write between the two?
> […](#)

Yeah Exactly That's what i want

Daniel-V-Richardson, 2024-03-31:
> Not sure why you need that. From BAC1, you can point directly to BAC0 (the port, not the app) using unicast (a direct connection). I see no need for an intermediate device.
>
> BBMD are used to make [...]

Yes I understand it's kinda unnecessary, but i just wanted to try it out as niagra does it pretty efficiently, and not only bacnet to banet, i also wanted to expose modbus points to be converted as Bacnet points from my application too.

closed 2024-06-07 by github-actions[bot]