ChristianTremblay/BAC0 #446: Device properties not accessible

jhughesbiot, 2024-02-29:
I have been struggling to find any real world way to resolve device properties.  I have successfully discovered and read all device properties with BAC0.lite and readMultiple but the response to that read is a structure of data that seems to have no relationship to what property these values are from.  I tried to enable show_property_name but since the device has proprietary properties, it fails.  

If I try to read just the protocol-version (other variations protocol_version and capitalized variants to no avail) singularly, it fails to resolve this.  Other BACnet explorers have no problem resolving these properties and this one specifically so what's the deal here?  Why can't I read a known required BACnet property of the device?  Why can't i get actual device property information that makes any sense to process?

Works but the response is unusable without property names or something that lets you process these to known property addresses/values:
`props = bacnet.readMultiple(f'{device[2]} device {device[3]} all',show_property_name=False)`

Absolute failure of a request to what should be a simple property read
```
object_identifier = f"{device[2]} device {device[3]} protocol-version"
version = bacnet.read(object_identifier, show_property_name=False)
```

> confirmed request encoding error: ValueError("undefined enumeration 'protocol-version'")
> Traceback (most recent call last):
>   File "C:\AppData\Local\Programs\Python\Python311\Lib\site-packages\bacpypes\appservice.py", line 1486, in sap_indication
>     apdu.encode(xpdu)
>   File "C:\AppData\Local\Programs\Python\Python311\Lib\site-packages\bacpypes\apdu.py", line 695, in encode
>     Sequence.encode(self, self._tag_list)
>   File "C:\AppData\Local\Programs\Python\Python311\Lib\site-packages\bacpypes\constructeddata.py", line 111, in encode
>     helper = element.klass(value)
>              ^^^^^^^^^^^^^^^^^^^^
>   File "C:\AppData\Local\Programs\Python\Python311\Lib\site-packages\bacpypes\primitivedata.py", line 1167, in __init__
>     raise ValueError("undefined enumeration '%s'" % (arg,))
> ValueError: undefined enumeration 'protocol-version'

ChristianTremblay, 2024-02-29:
in the pip version of BAC0, I use bacpypes and the name of that property is : `protocolVersion`
[ref : [bacpypes](https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py#L1630) ]

in the next version of BAC0, I will use bacpypes3 and this will now be `protocol-version`

```
# Sync BAC0 (actual version)
object_identifier = f"304:4 device 5404 protocolVersion"
version = bacnet.read(object_identifier)
print(version)
```

```
# Async BAC0 (next version)
object_identifier = f"304:4 device 5404 protocol-version"
version = await bacnet.read(object_identifier)
print(version)
```

Now I'm not sure what is your use case but please keep in mind that I made a tool for me first, which I then shared for the benefits of others. My goal was not to create another BACnet explorer tool. I needed something that would allow me to quickly connect to BACnet device to interact with them. Write to setpoints, read values, make overrides, simulate inputs, etc. 

Most of BACnet properties (eventState, maxValue, protocolVersion, etc...) are not something I work a lot with, so there is not a lot done for them yet.

you can read them using : 

```
dev['pointName'].bacnet_properties
```
to get a list, and it's possible to write back to them when supported.

I can understand that sometimes it may be frustrating not finding what we are looking for, that said, BAC0 is open source. If you really need something, I will gladly accept PR.

ChristianTremblay, 2024-03-30:
In BACnet, objects have "properties". For example, present-value, reliability, max-value, min-value, etc...
Each of those properties can be polled individually when required.

readMultiple used with all, simply means : read all properties from that objects.

Problem is, the list of properties can be long, the device you're interrogating may not support read multiple or worst, segmentation... and if you care about speed and traffic, you cannot update all properties for each point on each devices.

So reading "all" may be seen as a way to initialize an object, or explore it. But it is not something broadly used.

show_property_name, argument of readMultiple is typically not needed in my use case. Because I rarely use all. I choose only the properties I want to update in the requests I make. Which means that I know up front the names of the properties I'm updating.

But when calling a readMultiple using all, show_property_name should be true, because as you say, you just don't know what you are reading.

The results you see are coming directly from bacpypes. 

It is used in BAC0 as a convenience for exploration when you call `device['point'].bacnet_properties`

It's to differenciate bacnet_properties (those coming from network) with BAC0 point properties which are properties I created to document the points I'm working with inside a device.

ex. Those are BACnet properties coming from a JCI controller for OA-T (outside air temp)

```shell
{<PropertyIdentifier: 3369>: None,
 <PropertyIdentifier: 3113>: None,
 <PropertyIdentifier: 3371>: None,
 <PropertyIdentifier: 513>: None,
 <PropertyIdentifier: 663>: None,
 <PropertyIdentifier: units>: <EngineeringUnits: degrees-celsius>,
 <PropertyIdentifier: 661>: None,
 <PropertyIdentifier: 3376>: None,
 <PropertyIdentifier: 3930>: None,
 <PropertyIdentifier: present-value>: 20.0,
 <PropertyIdentifier: relinquish-default>: 13.0,
 <PropertyIdentifier: priority-array>: [<bacpypes3.basetypes.PriorityValue at 0x22821bf2390>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bf1a10>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bf24d0>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bdef10>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bdcd90>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bdd7d0>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bded50>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bdddd0>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bdefd0>,
  <bacpypes3.basetypes.PriorityValue at 0x22821b6b750>,
  <bacpypes3.basetypes.PriorityValue at 0x22821bdd9d0>,
  <bacpypes3.basetypes.PriorityValue at 0x22821b68a90>,
  <bacpypes3.basetypes.PriorityValue at 0x22821c50910>,
  <bacpypes3.basetypes.PriorityValue at 0x22821c53e90>,
  <bacpypes3.basetypes.PriorityValue at 0x22821c52c10>,
  <bacpypes3.basetypes.PriorityValue at 0x22821c51550>],
 <PropertyIdentifier: cov-increment>: 0.0010000000474974513,
 <PropertyIdentifier: event-detection-enable>: 0,
 <PropertyIdentifier: event-enable>: <EventTransitionBits: to-offnormal;to-fault;to-normal>,
 <PropertyIdentifier: limit-enable>: <LimitEnable: low-limit-enable;high-limit-enable>,
 <PropertyIdentifier: high-limit>: 70.0,
 <PropertyIdentifier: low-limit>: 65.0,
 <PropertyIdentifier: deadband>: 0.0,
 <PropertyIdentifier: time-delay>: 0,
 <PropertyIdentifier: notification-class>: 4194303,
 <PropertyIdentifier: notify-type>: <NotifyType: alarm>,
 <PropertyIdentifier: acked-transitions>: <EventTransitionBits: to-offnormal;to-fault;to-normal>,
 <PropertyIdentifier: event-time-stamps>: [<bacpypes3.basetypes.TimeStamp at 0x22821b9f710>,
  <bacpypes3.basetypes.TimeStamp at 0x22821a45050>,
  <bacpypes3.basetypes.TimeStamp at 0x22821bf2c50>],
 <PropertyIdentifier: event-message-texts>: ['', '', ''],
 <PropertyIdentifier: event-message-texts-config>: ['', '', ''],
 <PropertyIdentifier: 536>: None,
 <PropertyIdentifier: 32581>: None,
 <PropertyIdentifier: 32623>: None,
 <PropertyIdentifier: 4304>: None,
 <PropertyIdentifier: 4305>: None,
[... 21 lines cut ...]
 <PropertyIdentifier: event-state>: <EventState: normal>,
 <PropertyIdentifier: 32527>: None,
 <PropertyIdentifier: 2390>: None,
 <PropertyIdentifier: object-name>: 'OA-T',
 <PropertyIdentifier: description>: '',
 <PropertyIdentifier: 512>: None,
 <PropertyIdentifier: 673>: None,
 <PropertyIdentifier: 2197>: None,
 <PropertyIdentifier: 908>: None,
 <PropertyIdentifier: 1006>: None,
 <PropertyIdentifier: object-type>: <ObjectType: analog-value>,
 <PropertyIdentifier: object-identifier>: (<ObjectType: analog-value>, 62)}
```

And those are the BAC0's properties I keep track in the point object

```shell
{'device': CGM-21-005_AC-1 / Connected, 'name': 'OA-T', 'type': 'analog-value', 'address': '62', 'description': '', 'units_state': <EngineeringUnits: degrees-celsius>, 'simulated': (False, 0), 'overridden': (False, 0), 'priority_array': None, 'history_size': None, 'bacnet_properties': {<PropertyIdentifier: 3369>: None, <PropertyIdentifier: 3113>: None, <PropertyIdentifier: 3371>: None, <PropertyIdentifier: 513>: None, <PropertyIdentifier: 663>: None, <PropertyIdentifier: units>: <EngineeringUnits: degrees-celsius>, <PropertyIdentifier: 661>: None, <PropertyIdentifier: 3376>: None, <PropertyIdentifier: 3930>: None, <PropertyIdentifier: present-value>: 20.0, <PropertyIdentifier: relinquish-default>: 13.0, <PropertyIdentifier: priority-array>: [<bacpypes3.basetypes.PriorityValue object at 0x0000022821BF2390>, <bacpypes3.basetypes.PriorityValue object at 0x0000022821BF1A10>, [... cut ...]
As you can see, in the BAC0's properties you see bacnet properties BUT only because I called update_bacnet_properties which created a copy in BAC0's properties. It is not there by default (to reduce traffic).

Now you may wonder what does Property Identifier with numbers mean.

Those are "proprietary properties". They are defined by a vendor, the way they want, and they have no obligation to tell you what they are. There is no name inside proprietary properties... you need documentation from the vendor.

closed 2024-06-06 by github-actions[bot]