ChristianTremblay/BAC0 #448: Issue connecting to my devices during the Montreal daylight saving time transition period.

geckotronic, 2024-03-15:
I have a micro PC that connects to my controllers every 5 minutes, takes a temperature sample, and then disconnects. On March 10th, between 2 am and 3 pm, my script was giving an error when trying to connect. My micro PC is set to the America/Costa_Rica time zone, where daylight saving time changes do not apply. My controllers also have the daylight saving property disabled.

Upon reviewing the code, I found that the function shown below fails because it does not consider the time zone of my device; instead, it is hardcoded to the America/Montreal zone, where that hour does not exist due to daylight saving time.
[image]
[image]

Is there any way to force the library to use the America/Costa_Rica time zone when invoking BAC0.lite? Or perhaps in a new version, the time zone could be taken from the system?

Thank you very much.

ChristianTremblay, 2024-03-18:
Can you try

```python
bacnet = BAC0.lite()
bacnet.timehandler.set_timezone("America/Costa_Rica")
```

geckotronic, 2024-03-18:
Thanks Christian, 

I apologize if I wasn't clear before. I encounter the error when calling BAC0.lite() and cannot proceed to set the timezone. 

[image]

Here are the steps I'm following to reproduce the error:
- I set my computer's time zone to either UTC or America/Costa_Rica (a timezone without daylight saving).
- I set my computer's clock to 2:15 am on March 10, 2024.
- I attempt to initialize BAC0 with `bacnet = BAC0.lite()`.

I assume the failure occurs because, although my computer's time zone does not depend on daylight saving, initializing the library consider the America/Montreal zone, where that hour (2:15am 10March24) would not exist.

Is there a way to change the timezone before initializing the library, perhaps as a parameter at the start? Or could it possibly take the system's time zone?

If I change the initialization inside the library's code to America/Costa_Rica (inside TimeHandler class), the problem disappears, but I would prefer not to edit the library and instead make the change externally.

Thank you very much.

ChristianTremblay, 2024-03-18:
Can you test cloning the master branch and pip install it ? I just modified it to add the arg `timezone`

and use : 
```python
import BAC0
bacnet = BAC0.lite(timezone="America/Costa_Rica")
```

ChristianTremblay, 2024-03-18:
By the way, I've added that to the new async version that use bacpypes3 that you can try also. Look for branch `Async`

This will replace the actual version of BAC0 soon

geckotronic, 2024-03-18:
Omg!, thanks Christian. this works perfect!

closed 2024-03-18 by geckotronic