ChristianTremblay/BAC0 #449: BAC0 running in docker is not able to discover devices in network

shivangishakya, 2024-03-20:
I am not able to discover bacnet devices if runs BAC0 script in docker. It is working fine in local machine (windows). Tried: network_mode = host and creating a network with gateway as my router's IP.

ChristianTremblay, 2024-03-21:
This is a common issue with docker as UDP broadcast are not passing through...

Discovery will fail but I think you will be able to connect directly to a device (unicast)

shivangishakya, 2024-03-21:
Any way of bacnet discovery?

ChristianTremblay, 2024-03-22:
Using docker, it's hard and I haven't done a lot of tests.

Last time I worked on that, I found this : https://www.youtube.com/watch?v=bKFMS5C4CG0

But I haven't had time to fully test everything. Basically, it really is down to the UDP broadcast that need to travel to the container and from the container.

closed 2024-05-29 by github-actions[bot]