ChristianTremblay/BAC0 #454: How to disconnect only one device

DubarryXavier, 2024-04-30:
Hello, I'm trying to use BACnet to simulate 300 devices. However, I would like to disconnect some and reconnect them later. The issue is that when I disconnect one device, all the devices get disconnected. I'm using device_name.disconnect().
I tried assigning individual IP addresses to each device or giving them the same IP address. They all used different ports and IDs.

def disconnect_device(device_name):
    global devices
    if device_name in devices:
        device = devices[device_name]
        device.disconnect()  # Déconnexion du périphérique
        del devices[device_name]  # Suppression du périphérique de la liste des périphériques connectés
        print(f"Device {device_name} disconnected.")
    else:
        print(f"Device {device_name} not found.")
Thank you for your time

ChristianTremblay, 2024-05-11:
What if you don't delete it ?
and when you are ready to reconnect you use the old instance.... you `connect(network=bacnet)`

Have you tried the async branch ?

DubarryXavier, 2024-05-14:
Hello thanks for the answer, I will try the async branch.

closed 2024-05-14 by DubarryXavier