ChristianTremblay/BAC0 #458: Issue Accessing device visible

bdesalle, 2024-07-10:
Hi, 

I am having trouble accessing device that is well discovered due to no network ID. 

Overview of the infrastructure:
I have 
- a VM running on network 10.22.81.100/24  using Python 3.11.9 and BAC0 23.07.03
- a device on network 10.22.81.119/24 

My first goal is simply to access the device at 10.22.81.119 and read a data. Unfortunately I a stuck due to no BACnet network found. 

My code: 

import BAC0

**myIPAddr = '10.22.81.100/24'
bacnet = BAC0.connect(ip = myIPAddr)**
_2024-07-10 14:15:47,441 - INFO    | Starting BAC0 version 23.07.03 (Lite)
2024-07-10 14:15:47,442 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2024-07-10 14:15:47,442 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2024-07-10 14:15:47,442 - INFO    | Starting TaskManager
2024-07-10 14:15:47,443 - INFO    | Using ip : 10.22.81.100 on port 47808 | broadcast : 10.22.81.255
2024-07-10 14:15:47,457 - INFO    | Starting app...
2024-07-10 14:15:47,458 - INFO    | BAC0 started
2024-07-10 14:15:47,458 - INFO    | Registered as Simple BACnet/IP App
2024-07-10 14:15:47,458 - INFO    | Device instance (id) : 3056183
2024-07-10 14:15:47,459 - INFO    | Update Local COV Task started (required to support COV)_
**bacnet.whois()**
_[('10.22.81.119', 99)]_
**bacnet.devices**
_[('Canalis 11A', 'Universal Electric Corporation', '10.22.81.119', 99)]_
**bacnet.what_is_network_number()**
No data
**bacnet.discover()**
_2024-07-10 14:18:47,940 - INFO    | Found those networks : set()
2024-07-10 14:18:47,941 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
[('10.22.81.119', 99)]_

Your help is appreciated, I know I am missing something but for the love of god cannot find what. 

Thanks,

ChristianTremblay, 2024-07-15:
try to define the device directly : 
```
import BAC0

myIPAddr = '10.22.81.100/24'
bacnet = BAC0.connect(ip = myIPAddr)
BAC0.device('10.22.81.119', 99, bacnet)
```

bdesalle, 2024-07-15:
Hi, 

Thanks, it helped bit now it crashes 
>>> BAC0.device('10.22.81.120', 99, bacnet)
2024-07-15 19:32:04,978 - INFO    | Changing device state to DeviceDisconnected'>
2024-07-15 19:32:04,985 - INFO    | Changing device state to RPDeviceConnected'>
2024-07-15 19:32:04,994 - INFO    | Device 99:[Canalis 11A] found... building points list
2024-07-15 19:32:06,700 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2024-07-15 19:32:06,715 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
2024-07-15 19:32:06,730 - WARNING | The description property is not implemented in the device. Using a default value for internal needs.
.....
....
....
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/Device.py", line 195, in __init__
    self.new_state(DeviceDisconnected)
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/Device.py", line 211, in new_state
    self._init_state()
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/Device.py", line 857, in _init_state
    self.connect()
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/Device.py", line 900, in connect
    self.new_state(RPDeviceConnected)
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/Device.py", line 211, in new_state
    self._init_state()
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/Device.py", line 454, in _init_state
    self._buildPointList()
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/Device.py", line 541, in _buildPointList
    ) = self._discoverPoints(self.custom_object_list)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/mixins/read_mixin.py", line 194, in _discoverPoints
    self._process_new_objects(
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/mixins/read_mixin.py", line 389, in _process_new_objects
    presentValue = self.read_single(
                   ^^^^^^^^^^^^^^^^^
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/devices/mixins/read_mixin.py", line 661, in read_single
    return self.properties.network.read(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/BAC0/core/io/Read.py", line 162, in read
    value = apdu.propertyValue.cast_out(datatype)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/bacpypes/constructeddata.py", line 1346, in cast_out
    helper = klass(self.tagList[0])
             ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/bacpypes/primitivedata.py", line 1157, in __init__
    self.decode(arg)
  File "/root/.venv/python-3.11/lib/python3.11/site-packages/bacpypes/primitivedata.py", line 1238, in decode
    raise InvalidTag("enumerated application tag required")
bacpypes.errors.InvalidTag: enumerated application tag required

ChristianTremblay, 2024-07-16:
For old devices or devices that are not following all BACnet rules, you have to be selective.... 

You can read the object list property and keep only the ones you want to use.

https://bac0.readthedocs.io/en/latest/controller.html#object-list

You can also skip the "device"creation and use bacnet.read() or bacnet.write(). You will need to know the properties of the vlaues of interest.

bdesalle, 2024-07-16:
Hi Christian, 

Thanks a lot, I am now able to connect to the devices and read values. 

Just one last question, is there a way to export all object lists from the device ? 
In my device there is hundreds of object and most of them useless. And while it is possible using YABE connected to the network to get the list, I still have to manually check them all via the GUI. 

I just hope there is an easier way. 
Thanks a lot for your help

ChristianTremblay, 2024-07-19:
Sometimes, different versions of Python get mixed and we find missing packages...if pandas is installed in the same env than BAC0, I see no reason to trigger that exception

bdesalle, 2024-07-24:
Thank you for your help, 
As this now turns into an environnment issue, I am closing this issue

closed 2024-07-24 by bdesalle