ChristianTremblay/BAC0 #459: Doesn't work with Python 3.12 since asyncore is removed

yyli, 2024-07-17:
aysncore was removed thus `BACpypes` library can't be imported.

`BACpypes3` seems to move to asyncio and I monkey patched my use case for now. 

It seems you're already working on a `BACpypes3` version but just wanted to file an issue for visibility

https://github.com/ChristianTremblay/BAC0/issues/457#issuecomment-2217777934

ChristianTremblay, 2024-07-19:
https://github.com/ChristianTremblay/BAC0/tree/async

yyli, 2024-07-25:
Do you know how I can depend on this in my project. It is possible if you there’s no version published?

dechamps, 2024-08-04:
@yyli 

It's usually trivial to build and install a pip package yourself from source - just clone the repo, checkout the branch and then run `pip install /path/to/source/directory`.

For those who want to stick to officially released versions, the workaround for this is to create an isolated Python 3.11 venv and use BAC0 there.

ChristianTremblay, 2024-09-08:
I'm slowly preparing the pip version
BAC0 will now be async, based on BACpypes3
Users will need to migrate their code or stick with the Legacy tagged version

ChristianTremblay, 2024-09-10:
This is fixed. If you `pip install BAC0` now it will install the new version that works with BACpypes3, python 3.10+

closed 2024-09-10 by ChristianTremblay