ChristianTremblay/BAC0 #460: BAC0 on kubernetes

CSEGarazas, 2024-07-19:
Hi, I`m trying to run BAC0.lite() on kubernetes container 
But I can`t init BAC0 device.
I`m getting such log and error:

Initializing bacnet connection for worker.
2024-07-19 14:46:18,180 - INFO    | Starting BAC0 version 21.02.25 (Lite)
2024-07-19 14:46:18,180 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2024-07-19 14:46:18,180 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2024-07-19 14:46:18,181 - INFO    | Starting TaskManager
2024-07-19 14:46:18,181 - INFO    | Using ip : x.x.x.x:47810
2024-07-19 14:46:18,182 - ERROR   | an error has occurred: [Errno 99] Cannot assign requested address

Any ideas why BAC0 bind ip and port to kubernetes container?

ChristianTremblay, 2024-07-19:
When ip is not provided, it tries to reach google.com and bind to the interface that gives a response.

bbartling, 2024-07-20:
I don't know about kubernetes containers but aren't those meant for scaling
up cloud based apps? BACnet is an "intranet" local area network protocol
behind the fire wall... Do people use kubernetes for that? Or can you just
run in a docker container or Linux systemd service..?

CSEGarazas, 2024-07-21:
> When ip is not provided, it tries to reach google.com and bind to the interface that gives a response.

Instead x.x.x.x I`m using Kubernetes POD ip address . And I don`t know why but it can`t bind IP address. 
Priviously I was running my app in docker desktop. And BAC0 library work perfectly on container. But now we need scale our app by clients. And for this reason why would like to use Kuberntes services. I can`t find any reason why it`s not working ...

bbartling, 2024-07-21:
Are you running a BACnet app on the cloud and making BACnet requests across
the internet through VPN service??

CSEGarazas, 2024-07-22:
Firstly, I'm trying to initiate the BAC0 app, but I get an error during initialization. I tried to expose the port, but it didn't help. If I succeed in initiating the BAC0 app, I will send requests to the controller, which will be in a private VPN network.

ChristianTremblay, 2024-07-22:
Have you tried initializing BAC0 using the IP of the VPN interface ?

closed 2024-09-28 by github-actions[bot]