ChristianTremblay/BAC0 #462: History for Boolean/Enum points which have a propertyAccessError get registered with coerced values

ohshazbot, 2024-07-31:
The _trend implementation for BooleanPoint and EnumPoint does coercion of the valuu, this coercion doesn't consider the None case for these values. None values can occur if specify an object_list to the device which includes non-existant points (in my case, it was a once valid point that was later removed without my knowledge). Because of the translation happening, it inserted values into the history of the Point which were artificial, making me believe the point still existed. 

I beleive the _trend implementation for these two points needs to have their assignment to res be conditioned on res being not None.

ChristianTremblay, 2024-08-10:
Interesting use case. What do you suggest ?

ohshazbot, 2024-08-12:
Simplest solution, I think, would be updating the _trend implementations for Boolean & Enum Points to only replace the `res` value if `res is not None`

closed 2024-10-19 by github-actions[bot]