ChristianTremblay/BAC0 #463: Connect to Bacnet MSTP device on through router on another subnet

viraj-96, 2024-08-06:
Hi I have a bit of a weird situation. Im on a corporate network and my local machine is on a different subnet than the Bacnet router/controller(Trane SC+ if that helps). 

I am able to connect directly to the router and read points on it, however I ab unable to read points or connect to MSTP devices on the other side of the router that are visible on Yabe. Any ideas on how to make this work?

```
my_ip = '10.45.12.150'
bbmdIP = '10.45.52.20:47808'
bacnet = BAC0.connect(ip=my_ip, bbmdAddress=bbmdIP)
```

The discover function doesn't seem to find anything either.  
[image]

Connecting to Router works
[image]

Cant connect using MSTP formatting 
[image]

mirhamza708, 2024-08-07:
did you try this?
```
bacnet = BAC0.connect(ip='Your IP / your subnet mask', port='port', deviceId='some random device ID')
# BAC0.connect(ip='10.45.12.15/24', port='47808', deviceId='123654')
bacnet.discover()
```

viraj-96, 2024-08-07:
Was able to get this working on mu side. I had forgot to specify bbmdTTL in my initial connection, now I can see all devices. 
Thanks

closed 2024-08-07 by viraj-96