ChristianTremblay/BAC0 #471: RuntimeError: no running event loop

leaf188, 2024-09-11:
The following error message appears when you run BAC0.lite()
`
root@vm01:~# python
Python 3.12.0 (main, Oct 21 2023, 17:42:12) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import BAC0
>>> bacnet=BAC0.lite()
[09/11/24 07:54:41] INFO     2024-09-11 07:54:41,724 - INFO    | Starting Asynchronous BAC0 version 2024.09.10 (Lite)           notes.py:267
                    INFO     2024-09-11 07:54:41,728 - INFO    | Using bacpypes3 version 0.0.98                                 notes.py:267
                    INFO     2024-09-11 07:54:41,729 - INFO    | Use BAC0.log_level to adjust verbosity of the app.             notes.py:267
                    INFO     2024-09-11 07:54:41,730 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')       notes.py:267
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.12/dist-packages/BAC0/scripts/Lite.py", line 141, in __init__
    self._ping_task.start()
  File "/usr/local/lib/python3.12/dist-packages/BAC0/tasks/TaskManager.py", line 143, in start
    self.aio_task = asyncio.create_task(self.execute(), name=f"aio{self.name}")
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 407, in create_task
    loop = events.get_running_loop()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: no running event loop
`

ChristianTremblay, 2024-09-11:
You must run 

Python -m asyncio

This way, an event loop will be available. 

Then

import BAC0
bacnet = BAC0.start()

haithngn, 2024-09-23:
I am facing with the same issue on Mac M1, python 3.12.6 (via pyenv)

also tried,
```
python -m asyncio
asyncio REPL 3.12.6 (main, Sep  6 2024, 19:03:47) [Clang 15.0.0 (clang-1500.3.9.4)] on darwin
Use "await" directly instead of "asyncio.run()".
Type "help", "copyright", "credits" or "license" for more information.
>>> import asyncio
>>> import BAC0
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.12/3.12.6/Frameworks/Python.framework/Versions/3.12/lib/python3.12/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.12/3.12.6/Frameworks/Python.framework/Versions/3.12/lib/python3.12/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/opt/homebrew/Cellar/python@3.12/3.12.6/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/__main__.py", line 33, in callback
    coro = func()
           ^^^^^^
  File "<console>", line 1, in <module>
ModuleNotFoundError: No module named 'BAC0'
>>> 
```

@ChristianTremblay can you please have a look, do I need a Linux or Windows machine?

haithngn, 2024-09-23:
update,
I have switched to python 3.10.15, now BAC0 is working fine

haithngn, 2024-09-24:
hi guys, 
I have succeed read data from a BACnet device (Yabe), confirm BAC0 works fine.

@leaf188 , you have to place BAC0 instance somewhere that you keep it alive, for now, I have to put in into Flask middleware (for workaround).

thank you @ChristianTremblay, this issue can be closed I think.
and we can have a more detail example the perfect for developer who is not much familiar with Python.

haithngn, 2024-09-24:
p/s: the legacy versions 23.07.03 is still good, there is some news in asyncio version need to be learned before using.

tstorek, 2024-11-04:
@haithngn I having the same problem doing the transition to asyncio. With former versions I used Pyhton-Threading instead of asyncio. Could you please provide i Minimal working example (MWP) on how you did it. I need to run from PyCharm as well without Jupyter. Thanks

closed 2025-01-12 by github-actions[bot]