ChristianTremblay/BAC0 #472: RuntimeError "no broadcast" when attempting device discovery with defaults

mswank-tagup, 2024-09-11:
I've simply created a BACnet connection with the default options and issued a discovery (whois).

```
python -m asyncio

import BAC0
import ipaddress

bacnet_address_cidr="0.0.0.0/32"
bacnet_port=47808

ip_net = ipaddress.ip_network(bacnet_address_cidr)
bacnet = BAC0.lite(
    ip=format(ip_net.network_address, "s"),
    port=bacnet_port,
    mask=ip_net.prefixlen
)

_ = bacnet.discover(networks="known", limits=(0, 4194303), global_broadcast=False)
```

This is the bottom of the stacktrace:
```
/lib/python3.12/site-packages/bacpypes3/ipv4/__init__.py", line 241, in indication
    raise RuntimeError("no broadcast")
```

It appears that the underlying bacpypes3 application does not have a broadcast_address when it expects to. Weird since the BAC0 connection recognizes the proper broadcast addr: 
2024-09-11 14:43:34,321 - INFO    | Using ip : 0.0.0.0/32 on port 47808 | broadcast : 0.0.0.0

Is there a step that I'm missing? Patiently awaiting updated documentation :D

ChristianTremblay, 2024-09-11:
I don't understand the internal details of using 0.0.0.0/32

I'm in fact trying to understand this on another feature where another device sends messages to this placeholder instead of a standard broadcast address. 

For now you need to use a valid IP/Subnet mask for BAC0 to work.

ChristianTremblay, 2024-09-11:
Don't wait for doc !

Help me build them ;-)

ChristianTremblay, 2024-09-12:
I looked for hints about this, and I would really like you to explain what is your use case actually ? What is the advantage of using 0.0.0.0/32 as the ip address of your device instead of your actual network interface address ?

Thanks for the details

ChristianTremblay, 2024-09-13:
I made numerous test and went up the stack until I get to asyncio.DataGramProtocole

Even when binding to 0.0.0.0 I never get to receive datagrams

My conclusion (supported by @joelbender) is that there is probably an OS limitation somewhere, maybe related to security, or else preventing the use of 0.0.0.0 as destination address in datagram messages. 

To use BAC0, you need to provide a valid IP with a valid subnet that will create in bacpypes3.pdu.Address a valid broadcast address.

closed 2024-09-13 by ChristianTremblay