ChristianTremblay/BAC0 #478: Points not detected by YABE

SIDRAGNEEL, 2024-09-19:
Hello, after running the create device test code with a modified add point function, YABE doss not show any points but a Network port. Is this a code issue or a have issue ?
 [image]

[image]

ChristianTremblay, 2024-09-23:
The script you show me creates 2 bacnet devices. One on 0xBAC0, the other one on 0xBAC1
The one with all the points added is on port 47809 (0xBAC1)
The one you look at on YABE is the one created first in the script, which I use for connecting to devices when testing.

If you want to create 1 device, use only the first async with 

You should also remove time.sleep and maybe replace that with an asyncio.sleep or you will pause the execution of the script everytime you call sleep.

SIDRAGNEEL, 2024-09-26:
> The script you show me creates 2 bacnet devices. One on 0xBAC0, the other one on 0xBAC1
> The one with all the points added is on port 47809 (0xBAC1)
> The one you look at on YABE is the one created [...]

Thanks a lot. Is there a way to create two independent devices that  are from the same IP , perhaps using different ports, to read /write on the other devices ?

ChristianTremblay, 2024-09-26:
https://github.com/ChristianTremblay/BAC0/blob/main/tests/manual_test_create_device.py

closed 2024-12-03 by github-actions[bot]