ChristianTremblay/BAC0 #485: Read Multiple request_dict Returns One

chrisreichert53, 2024-09-27:
I believe there is a bug in the readMultiple method when providing request_dict. It looks like in each iteration of the response, it resets the request_dict[object_identifier] to an empty list. This works fine for single properties but only returns the last property when multiple are provided.

https://github.com/ChristianTremblay/BAC0/blob/ef972e31577ebbcb3df6983ad69bbfbcd52b4a9c/BAC0/core/io/Read.py#L297-L317

Suggestion in place of:

https://github.com/ChristianTremblay/BAC0/blob/ef972e31577ebbcb3df6983ad69bbfbcd52b4a9c/BAC0/core/io/Read.py#L317

to

```
if str(object_identifier) not in dict_values:
    dict_values[str(object_identifier)] = []
```

ChristianTremblay, 2024-10-12:
If you can test develop branch, your proposal is added

Thanks

ChristianTremblay, 2024-11-21:
https://github.com/ChristianTremblay/BAC0/issues/485

ChristianTremblay, 2024-11-21:
@bill-clapham please add details as I thought it was fixed

bill-clapham, 2024-11-22:
Hi Christian, thanks for looking in to this. 
In the main branch of BAC0 2024.9.20 the bug highlighted by @chrisreichert53  on line 317 of Read.py is what was causing my error.
for document continuity I will add my code extracts. 

Sample code
```
    bacnet = BAC0.start(BACNET_ADAPTER)
    _rpm = {'address': '192.168.2.100:47808',
              'objects': {'analogValue:14': ['objectName', 'presentValue', 'statusFlags', 'units'],
                          'analogValue:15': ['presentValue', 'objectName']}}

    readRes = await bacnet.readMultiple('192.168.2.100:47808', request_dict=_rpm)
    pprint(readRes)
```

Above code produced:
```
{'analog-value,14': [(<PropertyIdentifier: units>, <EngineeringUnits: degrees-celsius>)],
 'analog-value,15': [(<PropertyIdentifier: object-name>, "JesPl'CHEP_FCU_106'RHvacCoo'TCtlC'SpCPrt")]}
```

After making the suggested change to Read.py the same code produces:
```
{'analog-value,14': [(<PropertyIdentifier: object-name>, "JesPl'CHEP_FCU_106'RHvacCoo'TCtlC'SpCCmf"),
                     (<PropertyIdentifier: present-value>, 24.0),
                     (<PropertyIdentifier: status-flags>, <StatusFlags: >),
                     (<PropertyIdentifier: units>, <EngineeringUnits: degrees-celsius>)],
 'analog-value,15': [(<PropertyIdentifier: present-value>, 40.0),
                     (<PropertyIdentifier: object-name>, "JesPl'CHEP_FCU_106'RHvacCoo'TCtlC'SpCPrt")]}
```

The correct data is now shown but in a different structure as documented in https://bac0.readthedocs.io/en/latest/read.html

Using:
Python, Version 3.13
BAC0, Version: 2024.9.20
bacpypes3, Version: 0.0.98

Again, big thanks for your work Christian 👍👍

ChristianTremblay, 2024-12-05:
Have you tried the develop branch to see if the issue is fixed ?

bill-clapham, 2025-01-28:
Hi @ChristianTremblay Apologies for the delay, been offline recently. 
I tried the latest code from the developer branch (copied on 29 Jan) but have the same issue where the keys in the returned dict are strings instead of tuples and the values for each key is a list of tuples but the tuple items are bacpypes3 classes.

I'm an amateur programmer at best so please excuse any wording inaccuracies.
Regards
Bill

closed 2025-04-07 by github-actions[bot]