ChristianTremblay/BAC0 #493: Cannot write floating point values with _write()

skygreenslade, 2024-10-28:
I cannot write floating point values with _write(). Instead "null" is being written, and no error or return value is provided.

Example which creates the issue:
```
async with BAC0.start(ip=_ip) as bacnet:
        result = await bacnet._write(f'{_device_ip}:{_port} analogOutput 1 presentValue 0.1 - 3')
        print(f'\nResult: {result}\n')
```

Output (truncated):
```
Result: None
```

I think this may be due to the regex parsing the write arguments parsing a word for the value. Is there any workaround for this?

ChristianTremblay, 2024-11-11:
I'm pretty sure the write function doesn't return something. 

I just send this to the network

I highly suggest you to have wireshark open. Just to see what is going on on the network.

ChristianTremblay, 2024-11-21:
you were right on the regex. Thanks
Please try the develop branch

I also added the `return response` to the _write fucntion, tell me if this works for you

skygreenslade, 2024-11-27:
Thanks for the update! I am currently working on other projects, but I will follow up if I am able to test out the new functionality.

closed 2025-02-04 by github-actions[bot]