ChristianTremblay/BAC0 #495: Can't read device with new version while it works with 23.7.3

eloigabal, 2024-11-07:
We are using the Library to read from a few buildings and it does not work with the new version.

Here are the snippets we are using

```python
import asyncio
import BAC0

async def bac():
    bacnet_dev = {'deviceIp': '10.10.10.15',  'type': 'analogInput', 'objectId': '27'}
    print(BAC0.infos.__version__) # 2024.9.20
    async with BAC0.start('10.187.176.185/16', bbmdAddress='10.187.104.102:47808', bbmdTTL=900) as bacnet:
        # there is a routing to the external IP to the internal one
        try:
            value = await bacnet.read(f"{bacnet_dev['deviceIp']} {bacnet_dev['type']} {bacnet_dev['objectId']} presentValue")
        except Exception as e:
            print(f"can't read {attempt}/5: {e}")
asyncio.run(bac())
```

##### OUTPUT:
```sh
2024.09.10
2024-11-07 14:59:43,323 - INFO    | Starting Asynchronous BAC0 version 2024.09.10 (Lite)
2024-11-07 14:59:43,323 - INFO    | Using bacpypes3 version 0.0.98
2024-11-07 14:59:43,323 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2024-11-07 14:59:43,324 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2024-11-07 14:59:43,326 - INFO    | Using ip : 10.187.176.185/16 on port 47808 | broadcast : 10.187.255.255
2024-11-07 14:59:43,330 - INFO    | Using default JSON configuration file
2024-11-07 14:59:43,341 - INFO    | Registering as a foreign device to host 10.187.104.102:47808 for 900 seconds
2024-11-07 14:59:43,343 - INFO    | Registered as BACnet/IP App | mode foreign
2024-11-07 14:59:43,345 - INFO    | Device instance (id) : 3056980
2024-11-07 14:59:43,345 - INFO    | BAC0|3056980 connected. Entering context manager.
/Users/eloigabal/Developement/CIMNE/bacnet_ingestor/venv/lib/python3.10/site-packages/bacpypes3/vendor.py:111: UserWarning: object type 56 for vendor identifier 842 already registered: <class 'bacpypes3.local.networkport.NetworkPortObject'>
  warnings.warn(
/Users/eloigabal/Developement/CIMNE/bacnet_ingestor/venv/lib/python3.10/site-packages/bacpypes3/vendor.py:111: UserWarning: object type 8 for vendor identifier 842 already registered: <class 'bacpypes3.local.device.DeviceObject'>
  warnings.warn(
2024-11-07 14:59:43,347 - INFO    | Installing recurring task Ping Task (id:4509907648)
2024-11-07 14:59:46,360 - ERROR   | BAC0.scripts.Lite | Lite | BAC0.core.io.Read | Trouble with Iam... Response received from 10.10.10.15 = []
can't read: 
2024-11-07 14:59:46,370 - INFO    | Stopping all tasks
2024-11-07 14:59:46,370 - INFO    | Ok all tasks stopped
2024-11-07 14:59:46,376 - INFO    | BACnet stopped
2024-11-07 14:59:46,376 - INFO    | BAC0|3056980 disconnected. Exiting context manager.
Exception in callback IPv4DatagramServer.set_broadcast_transport_protocol(<IPv4Address 10.187.176.185>)(<Task cancell...nit__.py:150>>)
handle: <Handle IPv4DatagramServer.set_broadcast_transport_protocol(<IPv4Address 10.187.176.185>)(<Task cancell...nit__.py:150>>)>
Traceback (most recent call last):
  File "/Users/eloigabal/Developement/CIMNE/bacnet_ingestor/venv/lib/python3.10/site-packages/bacpypes3/ipv4/__init__.py", line 159, in retrying_create_datagram_endpoint
    return await loop.create_datagram_endpoint(
  File "/usr/local/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 1385, in create_datagram_endpoint
    raise exceptions[0]
  File "/usr/local/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 1369, in create_datagram_endpoint
    sock.bind(local_address)
OSError: [Errno 49] Can't assign requested address
```

while with the old version:
```python
def bac():
    bacnet_dev = {'deviceIp': '10.10.10.15', 'device_uri': 'https://icat.cat#measurement-INE-08363-ixon-bacnet-10.10.10.15-analogInput-27-PT15M', 'type': 'analogInput', 'objectId': '27'}
    print(BAC0.infos.__version__)
    bacnet = BAC0.lite('10.187.176.185/16', bbmdAddress='10.187.104.102:47808', bbmdTTL=900) 
    for attempt in range(5):
        try:
            value = bacnet.read(
                f"{bacnet_dev['deviceIp']} {bacnet_dev['type']} {bacnet_dev['objectId']} presentValue")
            print(value)
            break
        except Exception as e:
            print(f"can't read {attempt}/5")
    bacnet.disconnect()
bac()
```
##### OUTPUT
```bash
23.07.03
2024-11-07 15:02:56,521 - INFO    | Starting BAC0 version 23.07.03 (Lite)
2024-11-07 15:02:56,522 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2024-11-07 15:02:56,522 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2024-11-07 15:02:56,522 - INFO    | Starting TaskManager
2024-11-07 15:02:56,522 - INFO    | Using ip : 10.187.176.185 on port 47808 | broadcast : 10.187.255.255
2024-11-07 15:02:56,543 - INFO    | Starting app...
2024-11-07 15:02:56,543 - INFO    | BAC0 started
2024-11-07 15:02:56,543 - INFO    | Registered as Foreign Device
2024-11-07 15:02:56,544 - INFO    | Device instance (id) : 3056224
2024-11-07 15:02:56,545 - INFO    | Update Local COV Task started (required to support COV)
20.527509689331055
2024-11-07 15:02:56,986 - INFO    | Stopping all tasks
2024-11-07 15:02:57,540 - INFO    | Stopping TaskManager
2024-11-07 15:02:57,553 - INFO    | Ok all tasks stopped
2024-11-07 15:02:57,561 - INFO    | BACnet stopped
```

Are we doing something wrong?

ChristianTremblay, 2024-11-11:
I suspect bacpypes doesn't know which route to take to reach the device. 

Can you try a discover before the read ?

eloigabal, 2024-11-14:
Tank you for the fast response!

I tried the discover and in both versions, the new and the old one. Both versions does not find any devices. however the old can read the values if I explicitly specify the deviceIP.

I also noticed that both libraries trow an exception when I close the connection.
##### code BAC0-2024.9.20
```python
import asyncio
import BAC0

async def bac():
    bacnet_dev = {'deviceIp': '10.10.10.15', 'device_uri': '***', 'type': 'analogInput', 'objectId': '27'}
    print(BAC0.infos.__version__)
    async with BAC0.start('10.187.24.210/16', bbmdAddress='10.187.104.102:47808', bbmdTTL=900) as bacnet:
        await bacnet._discover()
        print(await bacnet.devices)
        try:
            value = await bacnet.read(
            f"{bacnet_dev['deviceIp']} {bacnet_dev['type']} {bacnet_dev['objectId']} presentValue")
        except Exception:
            print("ERROR: CANT READ DEVICE")
asyncio.run(bac())
```
##### output
```
2024.09.10
2024-11-14 10:46:31,208 - INFO    | Starting Asynchronous BAC0 version 2024.09.10 (Lite)
2024-11-14 10:46:31,209 - INFO    | Using bacpypes3 version 0.0.98
2024-11-14 10:46:31,209 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2024-11-14 10:46:31,209 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2024-11-14 10:46:31,236 - INFO    | Using ip : 10.187.24.210/16 on port 47808 | broadcast : 10.187.255.255
2024-11-14 10:46:31,252 - INFO    | Using default JSON configuration file
2024-11-14 10:46:31,261 - INFO    | Registering as a foreign device to host 10.187.104.102:47808 for 900 seconds
2024-11-14 10:46:31,262 - INFO    | Registered as BACnet/IP App | mode foreign
2024-11-14 10:46:31,264 - INFO    | Device instance (id) : 3056649
2024-11-14 10:46:31,264 - INFO    | BAC0|3056649 connected. Entering context manager.
2024-11-14 10:46:31,264 - INFO    | Installing recurring task Ping Task (id:4485000896)
2024-11-14 10:46:34,274 - WARNING | BAC0.scripts.Lite | Lite | BAC0.core.functions.Alias | Request timed out for what_is_network_number, no response
2024-11-14 10:46:38,277 - INFO    | Found those networks : set()
2024-11-14 10:46:38,278 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
2024-11-14 10:46:38,278 - INFO    | Issuing a local broadcast whois request.
2024-11-14 10:46:41,280 - INFO    | Discovery done. Found 0 devices on 0 BACnet networks.
None
2024-11-14 10:46:44,298 - ERROR   | BAC0.scripts.Lite | Lite | BAC0.core.io.Read | Trouble with Iam... Response received from 10.10.10.15 = []
ERROR: CANT READ DEVICE
2024-11-14 10:46:44,308 - INFO    | Stopping all tasks
2024-11-14 10:46:44,308 - INFO    | Ok all tasks stopped
2024-11-14 10:46:44,313 - INFO    | BACnet stopped
2024-11-14 10:46:44,313 - INFO    | BAC0|3056649 disconnected. Exiting context manager.
Exception in callback IPv4DatagramServer.set_broadcast_transport_protocol(<IPv4Address 10.187.24.210>)(<Task cancell...nit__.py:150>>)
handle: <Handle IPv4DatagramServer.set_broadcast_transport_protocol(<IPv4Address 10.187.24.210>)(<Task cancell...nit__.py:150>>)>
Traceback (most recent call last):
  File "/Users/eloigabal/Developement/CIMNE/Infraestructures/bacnet_ingestor/venv/lib/python3.10/site-packages/bacpypes3/ipv4/__init__.py", line 159, in retrying_create_datagram_endpoint
    return await loop.create_datagram_endpoint(
  File "/usr/local/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 1385, in create_datagram_endpoint
    raise exceptions[0]
  File "/usr/local/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 1369, in create_datagram_endpoint
    sock.bind(local_address)
OSError: [Errno 49] Can't assign requested address
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/Users/eloigabal/Developement/CIMNE/Infraestructures/bacnet_ingestor/venv/lib/python3.10/site-packages/bacpypes3/ipv4/__init__.py", line 167, in retrying_create_datagram_endpoint
    await asyncio.sleep(BACPYPES_ENDPOINT_RETRY_INTERVAL)
  File "/usr/local/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/tasks.py", line 605, in sleep
    return await future
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
[... 32 lines cut ...]
2024-11-14 10:50:03,269 - INFO    | Using ip : 10.187.24.210 on port 47808 | broadcast : 10.187.255.255
2024-11-14 10:50:03,282 - INFO    | Starting app...
2024-11-14 10:50:03,282 - INFO    | BAC0 started
2024-11-14 10:50:03,282 - INFO    | Registered as Foreign Device
2024-11-14 10:50:03,282 - INFO    | Device instance (id) : 3056614
2024-11-14 10:50:03,284 - INFO    | Update Local COV Task started (required to support COV)
2024-11-14 10:50:07,295 - INFO    | Found those networks : set()
2024-11-14 10:50:07,295 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
[]
18.717199325561523
2024-11-14 10:50:07,614 - INFO    | Stopping all tasks
2024-11-14 10:50:07,621 - INFO    | Stopping TaskManager
[... cut ...]
  File "/Users/eloigabal/Developement/CIMNE/Infraestructures/bacnet_ingestor/venv/lib/python3.10/site-packages/bacpypes/core.py", line 154, in run
    asyncore.loop(timeout=delta, count=1)
  File "/usr/local/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncore.py", line 214, in loop
    poll_fun(timeout, map)
  File "/usr/local/Cellar/python@3.10/3.10.14_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncore.py", line 151, in poll
    r, w, e = select.select(r, w, e, timeout)
OSError: [Errno 9] Bad file descriptor

```

rob2791, 2024-11-15:
Following, I'm having a similar problem (to my knowledge).

ChristianTremblay, 2024-11-21:
BACpypes3 is tied more closely to the BACnet standard by default so "weird routes" and devices not reachable via normal BACnet mechanism are harder to reach with default options.

First thing I see (that I missed when looking at the code on the phone) is that your network interface IP address is in the same range than the BBMD.

You should not need to register to a BBMD on the same subnet than you. The BBMD will handle your messages and send them where they should go. Normally.

Can you confirm that the BBMD device `10.187.104.102` talks to the `10.10.10.0/?` BBMD device (there should be a BBMD on that subnet).

If so, please try without registering to the BBMD­.

```python
import asyncio
import BAC0

async def bac():
    bacnet_dev = {'deviceIp': '10.10.10.15',  'type': 'analogInput', 'objectId': '27'}
    print(BAC0.infos.__version__) # 2024.9.20
    async with BAC0.start(ip='10.187.176.185/16') as bacnet:
        # there is a routing to the external IP to the internal one
        try:
            value = await bacnet.read(f"{bacnet_dev['deviceIp']} {bacnet_dev['type']} {bacnet_dev['objectId']} presentValue")
        except Exception as e:
            print(f"can't read {attempt}/5: {e}")
asyncio.run(bac())
```

Let start with this...

rob2791, 2024-12-27:
I'd like to pick this back up, as I've been unsuccessful at getting anything to work after updating (3 months ago). I've tried everything I can think of, reinstalling BAC0, different python versions, but I can't get anything to work.
I ran your simplified script and here is the output.  I'm not getting any response from the device. But I know the device works, Yabe shows the output without any issue. It is a different subnet obviously, but before the update everything worked just fine. I even added 47808 tcp/udp firewall rules just in case. 10.9.x.x is the server i'm running it from. 192.168.51.x is one of the bacnet devices (a generator). 

One interesting thing i saw is that the reported version is "2024.09.10", but I have "2024.9.20" installed. Not sure if it's a bug in reporting the version?

```

#!/usr/bin/python3.11
import asyncio
import BAC0
BAC0.log_level('debug')
async def bac():
    bacnet_dev = {'deviceIp': '192.168.51.xx', 'type': 'analogInput', 'objectId': '95'}
    print(BAC0.infos.__version__) # 2024.9.20
    async with BAC0.start() as bacnet:
        # there is a routing to the external IP to the internal one
        try:
            value = await bacnet.read(f"{bacnet_dev['deviceIp']} {bacnet_dev['type']} {bacnet_dev['objectId']} presentValue")
        except Exception as e:
            print(f"can't read {attempt}/5: {e}")
#            print(f"can't read")

asyncio.run(bac())
```

[image]
[image]
[image]

eloigabal, 2024-12-27:
Hello. We stoped reporting it as we got it running with my initial script.

I was working with a mac m1 and could not get it working as we are connecting to the network with a TAP openvpn that is not supoorted. My collegue wirh a linux system achieved to run it as the connection was done correctly.

i dont know if this can give you anny insigth on your problem! 

Best regards

ChristianTremblay, 2024-12-28:
The new flavour of bacpypes3 is really more sensitive to network configuration. The problem you have actually is that BAC0 tries to do a whois on the device in another subnet. A whois is a broadcast message typically, so if there is no BBMD between both networks, the message will never get an answer... hence the `[]` received.

I may have an idea...

I'll need a little time to investigate

ChristianTremblay, 2024-12-28:
I made some tests on my side (using the develop branch, maybe you can give it a try).
The whois generated is a unicast, if the network is set up correctly, the device should generate an answer and you should get it.

rob2791, 2024-12-28:
Thank you for taking the time.
It doesn’t look like the develop branch has been updated for a couple weeks. Is that the right branch you want me to try?

ChristianTremblay, 2024-12-28:
Yes, it is still ahead of what you find on pypi. 

Also, launch Wireshark and see what is going on.

rob2791, 2024-12-28:
I uninstalled Bac0 and did `pip3.11 install https://github.com/ChristianTremblay/BAC0/tarball/develop` but the results didn't change.
Here is packet capture and results of develop branch (which still says it's 2024.09.10?) 
[image]
[image]

Just to show you when i run a similar script on python3.9 here is the (correct) response..
[image]
[image]

Hopefully this is helpful..

ChristianTremblay, 2025-01-21:
From what I'm seeing on my side is that bacpypes3 is more strict on following BACnet rules regarding subnets.

Also, I added a retry mechanism for IAm in more recent develop version...if you can have a look... maybe we'll find something

closed 2025-03-30 by github-actions[bot]

nishad1092, 2026-03-15:
Hi, 
I'm having the similar issue. It works for few hours and then it stops and now I run it never discovers the device. This is the output I get:

Starting BACnet stack...
2026-03-15 12:03:46,000 - INFO    | Starting Asynchronous BAC0 version 2025.09.15 (Lite)
2026-03-15 12:03:46,000 - INFO    | Using bacpypes3 version 0.0.104
2026-03-15 12:03:46,000 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2026-03-15 12:03:46,000 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2026-03-15 12:03:46,014 - INFO    | Using ip : 192.168.178.122/24 on port 47808 | broadcast : 192.168.178.255
2026-03-15 12:03:46,054 - INFO    | Using default JSON configuration file
2026-03-15 12:03:46,061 - INFO    | Registered as BACnet/IP App | mode normal
2026-03-15 12:03:46,062 - INFO    | Device instance (id) : 3056709
2026-03-15 12:03:46,062 - INFO    | Installing recurring task Ping Registered Devices Task (id:4332293488)
2026-03-15 12:03:46,064 - INFO    | Installing recurring task Cleanup Tasks List (id:4329562320)
2026-03-15 12:03:47,063 - INFO    | BAC0|3056709 connected. Entering context manager.
Discovering devices...
2026-03-15 12:03:50,069 - WARNING | BAC0.scripts.Lite | Lite | BAC0.core.functions.Alias | Request timed out for what_is_network_number, no response
2026-03-15 12:03:54,071 - INFO    | Found those networks : set()
2026-03-15 12:03:54,072 - INFO    | Global braodacast required, attempting a simple whois using provided device instances limits (0 - 4194303)
2026-03-15 12:03:54,072 - WARNING | Issuing a global Broadcast whois can create network flood. Use with care.
2026-03-15 12:03:57,073 - INFO    | Discovery done. Found 0 devices on 0 BACnet networks.
_discover found nothing, trying who_is...
No devices found.
2026-03-15 12:04:10,078 - INFO    | Stopping all tasks
2026-03-15 12:04:10,078 - INFO    | Ok all tasks stopped
2026-03-15 12:04:10,080 - INFO    | BACnet stopped
2026-03-15 12:04:10,081 - INFO    | BAC0|3056709 disconnected. Exiting context manager.

Any help would be nice. I'm using the latest release: BAC0 2025.9.15. The device I'm simulating had device ID: 101, so why is it taking Device Instance: 305xxx. And why its not working, This discover is working only when I force the communication otherwise not. 

A snipet of my code:

async def discover_devices(bacnet):
    """Try _discover first, fall back to who_is. Returns list of (device_id, device_addr)."""

    # ── Method 1: _discover ───────────────────────────────────────────
    try:
        await bacnet._discover(global_broadcast=True)
        await asyncio.sleep(5)
        devices = bacnet.discoveredDevices
        if devices:
            result = []
            for key in devices:
                device_id   = int(key[0])
                device_addr = str(key[1]).split(":")[0]
                result.append((device_id, device_addr))
            print(f"Found {len(result)} device(s) via _discover.")
            return result
    except Exception:
        pass

    # ── Method 2: who_is fallback ───────────────────────────────────
    print("_discover found nothing, trying who_is...")
    try:
        iam_results = await bacnet.who_is()
        await asyncio.sleep(5)
        if iam_results:
            result = []
            for iam in iam_results:
                device_id   = int(iam.iAmDeviceIdentifier[1])
                device_addr = str(iam.pduSource).split(":")[0]
                result.append((device_id, device_addr))
            print(f"Found {len(result)} device(s) via who_is.")
            return result
    except Exception:
        pass

    return []

This worked well for few hours, it is running in my home network. Firewalls are disabled, ssh, udp port all are open. Even opened the NETbios in my router to not blacklist IP.

I suspect because of too many whois() broadcast, it congested te network to that specific IP and port. But it shouldn't happen right.  

What could be the issue ? How to resolve this ?