ChristianTremblay/BAC0 #498: BACnet Write Property Request Not Being Sent (Verified via Wireshark)

tjarkst, 2024-11-14:
The write_property method is not sending BACnet packets as expected. While read operations work correctly, write operations fail silently with no packets visible in Wireshark.

### Reproduction Steps
```python
import BAC0

# Initialize connection
bacnet = BAC0.lite(ip='192.168.0.97', port=port)

# Read works
read_result = bacnet.read('192.168.0.100 analogValue 459 presentValue')
print(f"Read successful: {read_result}")

# Write fails silently
write_result = bacnet.write('192.168.0.100 analogValue 459 presentValue 100 - 8')
print(f"Write attempted: {write_result}")

![image](https://github.com/user-attachments/assets/26cca76a-5016-4a06-b987-7c2cebeae650)

ChristianTremblay, 2024-11-14:
Try 

await bacnet._write(args)

And tell me if it is better

ChristianTremblay, 2024-11-14:
Complément : the write method creates a asyncio task. With await _write you force the write to happen now.

tjarkst, 2024-11-14:
It worked indeed

ChristianTremblay, 2024-11-14:
This is strongly related to "me learning asyncio" and trying to keep some function synchronous to be used in the REPL... I suspect at some point I'll need to clean that up. And rely on awaitable functions.

wdfiller, 2025-01-19:
I'm having this same issue of not being able to write the present value of an analog value.  I used the previous version of bacnet for a long time before it was updated to async and it worked great.  Can you possible give any more detail on a workaround for this bug? Or a full working example?  I've tried every combination and it either fails silently, gives a error of NoneType can't be used in 'await' expression or and number of arguments error.

tgc1979, 2025-02-08:
Hi

This does not solve the issue but hopefully can help with troubleshoot

I think  the reasions write () is not working is because it is passed to DoOnce() method which initialises a write task and appends to Task list where the recurring (Ping task) and one shot tasks are in the scope of asynch execute() should be executed  under TaskManager.py 

However

Async execute method checks various conditions like 
```
   if self.delay > 0:
            self.log(
                f"Installing recurring task {self.name} (id:{self.id})", level="info"
            )
```
But  DoOnce() does not populate self.dealy

I have added a default value in DoOnce() __init__ and set self.id and managed to move a little but still not working. 
And finally  gave up and started to use

`            await gateway._write('192.168.56.101 analog-output 10 presentValue 50 - 8')
`

ChristianTremblay, 2025-02-09:
A flaw in the way I approached async. 

I thought it was better to make synchronous functions that were calling task... so we don't have to "await" everything. it ends up being more complex... it is my first experience ever with async and await. 

I guess eventually, I will just make _write -> write

What do you think ?

wdfiller, 2025-02-20:
> Version *2025.02-18*
>
> Encountered this issue with multipleWrite(...) and write(...), too. Went
> to *Write.py* and saw that the former is commented out, and the latter
> fails in DoOnce part, so the task is not created. Used _write in the end;
> it worked.
>
> @ChristianTremblay <https://github.com/ChristianTremblay> I guess doing _write
> -> write and making write asynchronous makes sense, if _write is truly
> async (as in "non-blocking"). I see it awaits write_property, which in
> turn awaits request(write_property_request)...

You did a great job on this.  It allowed me to use the Maker edition of
Inductive Automation Ignition at home with a BACnet thermostat.  The BACnet
module is one of the few things that is not enabled on Maker Edition.

David Filler
SCADA Engineer, BSME
Yukon, OK

ChristianTremblay, 2025-02-21:
So..... I finally found some time to fix this. Thanks to @tgc1979 you were right, the problem was with the DoOnce function.

Have a look at the develop branch... this will soon be push up to master as this problem constantly reappears in the issue list.

closed 2025-04-30 by github-actions[bot]