ChristianTremblay/BAC0 #501: bacnet read property method return string instead list while reading PropertyList of an object

Sharath-Kumar-MR, 2024-12-03:
Hello Team, @ChristianTremblay 

BAC0 - 22.9.21
Python - 3.8.10

I am trying get properties list of a device like bacnet_client.read(f"{str(mac)} {str(bac_object)} {str(instance)} propertyList"), response of this should be a list instead it returns a string. 
expected response is ['eventState', 'outOfService', 'presentValue', 'reliability', 'statusFlags']
actual response is 'AutoIdealTemperature', I am not able to reproduce this case, its very random.
Could you pls let me know whats happening here.
Thank you.

Sharath-Kumar-MR, 2024-12-03:
2024-11-25 15:53:40,131 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    
2024-11-25 15:53:40,131 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ###################################
2024-11-25 15:53:40,131 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    # Read property
2024-11-25 15:53:40,131 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ###################################
2024-11-25 15:53:40,131 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ['4:109', 'analogValue', '401', 'propertyList']
2024-11-25 15:53:40,131 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ###################################
2024-11-25 15:53:40,132 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    REQUEST              <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x7715b24e5430>
2024-11-25 15:53:40,132 [bacpypes.task] DEBUG:    FunctionTask <bound method IOCB.abort of <bacpypes.iocb.IOCB(1712) instance at 0x7715b1d4b670>> (RuntimeError('timeout'),) {}
2024-11-25 15:53:40,132 [bacpypes.task] DEBUG:        - task: <bacpypes.task.FunctionTask.<locals>._FunctionTask object at 0x7715b1d0cf70>
2024-11-25 15:53:40,132 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    iocb                 <bacpypes.iocb.IOCB(1712) instance at 0x7715b1d4b670>
2024-11-25 15:53:40,133 [bacpypes.iocb._statelog] DEBUG:    1420.133217 4:109 active
2024-11-25 15:53:40,636 [bacpypes.iocb._statelog] DEBUG:    1420.636855 4:109 idle
2024-11-25 15:53:40,637 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    value                datatype            
2024-11-25 15:53:40,637 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    'AutoIdealTemperature' <class 'bacpypes.primitivedata.CharacterString'>

2024-11-25 15:26:41,999 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ###################################
2024-11-25 15:26:41,999 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    # Read property
2024-11-25 15:26:41,999 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ###################################
2024-11-25 15:26:41,999 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ['4:27', 'analogInput', '10', 'propertyList']
2024-11-25 15:26:41,999 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ###################################
2024-11-25 15:26:41,999 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    REQUEST              <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x7715b1d31a60>
2024-11-25 15:26:42,000 [bacpypes.task] DEBUG:    FunctionTask <bound method IOCB.abort of <bacpypes.iocb.IOCB(1514) instance at 0x7715b1d1f520>> (RuntimeError('timeout'),) {}
2024-11-25 15:26:42,000 [bacpypes.task] DEBUG:        - task: <bacpypes.task.FunctionTask.<locals>._FunctionTask object at 0x7715b1de2280>
2024-11-25 15:26:42,000 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    iocb                 <bacpypes.iocb.IOCB(1514) instance at 0x7715b1d1f520>
2024-11-25 15:26:42,000 [bacpypes.iocb._statelog] DEBUG:    3402.000906 4:27 idle
2024-11-25 15:26:42,001 [bacpypes.iocb._statelog] DEBUG:    3402.001171 4:27 active
2024-11-25 15:26:42,142 [BAC0_Root.BAC0.tasks.UpdateCOV.Update_local_COV] DEBUG:    Executing : update_local_cov_Update Local COV Task
2024-11-25 15:26:42,143 [BAC0_Root.BAC0.tasks.TaskManager.Manager] DEBUG:    Task 130935062743168 | update_local_cov_Update Local COV Task executed. update_local_cov_Update Local COV Task   | Avg exec delay : 5.01 sec | Avg latency : 0.01 sec | last executed : 2024-11-25 15:26:42 | Next Time : 2024-11-25 15:26:47
2024-11-25 15:26:42,411 [bacpypes.iocb._statelog] DEBUG:    3402.411922 4:27 idle
2024-11-25 15:26:42,412 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    value                datatype            
2024-11-25 15:26:42,413 [BAC0_Root.BAC0.scripts.Lite.Lite] DEBUG:    ['eventState', 'outOfService', 'presentValue', 'reliability', 'statusFlags', 'units'] <class 'bacpypes.constructeddata.ArrayOf.<locals>.ArrayOf'>

ChristianTremblay, 2024-12-17:
Looks like a bad or weird implementation of the BACnet device... AutoIdealTemperature is not something I know

ChristianTremblay, 2025-01-21:
Have you tried the latest version of BAC0 ?

closed 2025-03-30 by github-actions[bot]

Sharath-Kumar-MR, 2025-03-31:
We have not tried latest version of BAC0, got this handled with exceptions as of now, thank you