ChristianTremblay/BAC0 #504: Problem with write_weeklySchedule method (Any.encode() missing 1 required positional argument: 'self')

jucajata, 2024-12-08:
While I was trying to overwrite one of my weekly Schedule I found and issue with the parametrization of the Any instance.

The code is:
await bacnet_network.write_weeklySchedule(destination, schedule_instance, schedule)

destination: '10.xxx.xxx.xx'
schedule_instance: 2 (because the object name is "schedule 2")
schedule: {'states': {'inactive': 0, 'active': 1}, 'week': {'monday': [('9:45', 'active'), ('17:45', 'inactive')], 'tuesday': [('9:45', 'active'), ('17:45', 'inactive')], 'wednesday': [('9:45', 'active'), ('17:45', 'inactive')], 'thursday': [('9:45', 'active'), ('17:45', 'inactive')], 'friday': [('9:45', 'active'), ('17:45', 'inactive')], 'saturday': [('9:45', 'active'), ('17:45', 'inactive')], 'sunday': [('9:45', 'active'), ('17:45', 'inactive')]}}

How could I solve this issue with the BAC0==2024.9.20 version?

The complete Traceback:

Any.encode() missing 1 required positional argument: 'self'
2024-12-08 17:44:20,238 - INFO    | Installing recurring task Ping Task (id:2180480269680)
Exception in callback IPv4DatagramServer.set_local_transport_protocol(<IPv4Address 10.4.54.16>)(<Task cancell...nit__.py:150>>)
handle: <Handle IPv4DatagramServer.set_local_transport_protocol(<IPv4Address 10.4.54.16>)(<Task cancell...nit__.py:150>>)>  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
, in set_local_transport_protocol
    transport, protocol = task.result()
                          ^^^^^^^^^^^^^
  File "c:\Users\EFICIENCIACOLOMBIA\Documents\dev\GY_EE_BAC0\.venv\Lib\site-packages\bacpypes3\ipv4\__init__.py", line 159, in retrying_create_datagram_endpoint
    return await loop.create_datagram_endpoint(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\base_events.py", line 1445, in create_datagram_endpoint
    await waiter

ChristianTremblay, 2024-12-17:
Thanks for pointing that out, I'll have alook

ChristianTremblay, 2025-01-21:
Working on it...

ChristianTremblay, 2025-01-21:
Please try the develop branch

jucajata, 2025-01-25:
Hi Christian, I have tried to use the develop branch with the changes made, it does not generate any error and I get a message saying that the request has been sent to the controller to make the time change. I have sent the input parameters as they appear in the documentation, with the binary dictionary, and the days of the week. However, despite it saying that everything is correct, the changes are not reflected in the controller.

ChristianTremblay, 2025-01-27:
Thanks for the feedback... I'll dig a little more

ChristianTremblay, 2025-01-28:
I cannot replicate the issue

```
>>> hor = await bacnet.read_weeklySchedule('3:7', 10001)
>>> hor
{
    'object_references': [(<ObjectType: multi-state-value>, 10381)],
    'references_names': ['OCC-SCHEDULE'],
    'states': {'Occupied': 1, 'UnOccupied': 2, 'Standby': 3, 'Not Set': 4},
    'reliability': <Reliability: no-fault-detected>,
    'priority': 15,
    'presentValue': 'Occupied (1)',
    'week': {
        'monday': [('00:00', 'UnOccupied'), ('07:00', 'Occupied'), ('20:00', 'UnOccupied')],
        'tuesday': [('00:00', 'UnOccupied'), ('06:00', 'Occupied'), ('17:00', 'UnOccupied')],
        'wednesday': [('00:00', 'UnOccupied'), ('06:00', 'Occupied'), ('17:00', 'UnOccupied')],
        'thursday': [('00:00', 'UnOccupied'), ('06:00', 'Occupied'), ('17:00', 'UnOccupied')],
        'friday': [('00:00', 'UnOccupied'), ('06:00', 'Occupied'), ('17:00', 'UnOccupied')],
        'saturday': [('00:00', 'UnOccupied'), ('06:00', 'Occupied'), ('17:00', 'UnOccupied')],
        'sunday': [('00:00', 'UnOccupied'), ('06:00', 'Occupied'), ('17:00', 'UnOccupied')]
    }
}

>>> from BAC0.core.functions.Schedule import Schedule
>>> bacnet.write_weeklySchedule('3:7', 10001, hor2)    
>>> hor_b = await bacnet.read_weeklySchedule('3:7', 10001)  
>>> hor_b                                                                                                                                     
{                                                                                                                                             
    'object_references': [(<ObjectType: multi-state-value>, 10381)],                                                                          
    'references_names': ['OCC-SCHEDULE'],                                                                                                     
    'states': {'Occupied': 1, 'UnOccupied': 2, 'Standby': 3, 'Not Set': 4},                                                                   
    'reliability': <Reliability: configuration-error>,
    'priority': 15,
    'presentValue': 'Not Set (0)',
    'week': {
    'week': {
    'week': {
    'week': {
        'monday': [('01:00', 'Not Set'), ('17:00', 'Occupied')],
        'tuesday': [('02:00', 'Not Set'), ('17:00', 'Occupied')],
        'wednesday': [('03:00', 'Not Set'), ('17:00', 'Occupied')],
        'thursday': [('04:00', 'Not Set'), ('17:00', 'Occupied')],
        'friday': [('05:00', 'Not Set'), ('17:00', 'Occupied')],
        'saturday': [('06:00', 'Not Set'), ('17:00', 'Occupied')],
        'sunday': [('07:00', 'Not Set'), ('17:00', 'Occupied')]
    }
}
```

As you can see here I made it work on a JCI controller here. Can you give me more info about your issue ?

ChristianTremblay, 2025-01-28:
wait... I overlooked the configuration error...

ChristianTremblay, 2025-01-28:
Fixed https://github.com/ChristianTremblay/BAC0/commit/913f7c90182151f9510fdb1010c730dc052e6529
Can you try this ?

ChristianTremblay, 2025-01-29:
Also ! don't use await for the write !
This function is synchronous... (yeah I know, it's hard to figure out...)

jucajata, 2025-01-30:
This is all the code I have:

import BAC0
import asyncio
import socket
import psutil
from datetime import datetime
import calendar
import warnings
from sqlalchemy import create_engine, text
from dotenv import load_dotenv
import os
import sys

warnings.filterwarnings("ignore")  # Ignorar advertencias irrelevantes

# Cargar variables de entorno desde un archivo .env
load_dotenv()

class BACnetScheduler:
    def __init__(self, remote_ip, bbmdIP, bbmdTTL, port):
        self.remote_ip = remote_ip  # Nueva columna incluida
        self.bbmdIP = bbmdIP
        self.bbmdTTL = bbmdTTL
        self.port = port
        self.current_datetime = datetime.now().strftime('%Y-%m-%d_%H-%M-%S')
        self.my_ip = f"{self.obtener_ip_privada_10()[0]}/24"
        self.data_records = []

    def obtener_ip_privada_10(self):
        ips_privadas = []
        for _, direcciones in psutil.net_if_addrs().items():
            for direccion in direcciones:
                if direccion.family == socket.AF_INET and direccion.address.startswith('10.'):
                    ips_privadas.append(direccion.address)
        if ips_privadas:
            return ips_privadas
        else:
            raise ValueError("No se encontró ninguna IP privada en el rango 10.x.x.x")

    async def run_bacnet(self):
        try:
            # Conexión a la red BACnet con BBMD usando un puerto específico
            bacnet_network = BAC0.connect(ip=self.my_ip, bbmdAddress=self.bbmdIP, bbmdTTL=self.bbmdTTL, port=self.port)

            # Realiza un descubrimiento de dispositivos
            devices = await bacnet_network.who_is()
            if not devices:
                print(f"No se encontraron dispositivos en la red con IP {self.remote_ip}.")
                return

            # Itera sobre los dispositivos y extrae detalles
            for device in devices:
                try:
                    device_id = device.iAmDeviceIdentifier[1]
                    pdu_source = str(device.pduSource)

                    # Leer el nombre del dispositivo, si está disponible
                    try:
                        device_name = await bacnet_network.read(f"{pdu_source} device {device_id} objectName")
                    except Exception:
                        device_name = None

                    # Lee la lista de objetos del dispositivo
                    try:
                        object_list = await bacnet_network.read(f"{pdu_source} device {device_id} objectList")

                        # Filtra objetos Schedule comparando con el tipo de objeto
                        schedule_objects = []
                        for obj in object_list:
[... 217 lines cut ...]
2025-01-29 21:26:09,575 - INFO    | **Schedule Write request sent to device : 10.129.61.141**
Horario actualizado exitosamente en 2 para 10.129.61.141:47808.
2025-01-29 21:28:32,316 - INFO    | Installing recurring task Ping Task (id:1320985105360)
Exception in callback IPv4DatagramServer.set_local_transport_protocol(<IPv4Address 10.4.54.26>)(<Task cancell...nit__.py:150>>)
handle: <Handle IPv4DatagramServer.set_local_transport_protocol(<IPv4Address 10.4.54.26>)(<Task cancell...nit__.py:150>>)>
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "c:\Users\EFICIENCIACOLOMBIA\Documents\dev\GY_EE_BAC0\.venv\Lib\site-packages\bacpypes3\ipv4\__init__.py", line 176, in set_local_transport_protocol
    transport, protocol = task.result()
                          ^^^^^^^^^^^^^
  File "c:\Users\EFICIENCIACOLOMBIA\Documents\dev\GY_EE_BAC0\.venv\Lib\site-packages\bacpypes3\ipv4\__init__.py", line 159, in retrying_create_datagram_endpoint
    return await loop.create_datagram_endpoint(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\base_events.py", line 1445, in create_datagram_endpoint
    await waiter
asyncio.exceptions.CancelledError

---------------------------------------

But the Schedule is the same. What I'm doing wrong? I changed this line of code:

return Enumerated(dict_schedule["states"][v] - 1) 
with 
return Enumerated(dict_schedule["states"][v])

jucajata, 2025-01-30:
If you want, we can make a call

closed 2025-02-18 by ChristianTremblay

jucajata, 2025-03-24:
Thanks for all the support, the code works very good.