ChristianTremblay/BAC0 #515: Exception handling mismatch

CoMPaTech, 2025-01-21:
Just a quick note (might do a PR against it) - colleague just ran into this: https://github.com/ChristianTremblay/BAC0/blob/ef972e31577ebbcb3df6983ad69bbfbcd52b4a9c/BAC0/core/io/Write.py#L116

The exception won't work since it's incepting itself with `response` (which is only valid if the `try:` is successful, so this will never be reached). And yes this was obviously done through awaiting `_write()` instead of properly consuming `write()` without expecting an answer, but the errors generated would be the same obviously.

ChristianTremblay, 2025-01-21:
oups

CoMPaTech, 2025-01-21:
Willing to correct it - but didn't have time to start writing a PR just yet :)

closed 2025-03-30 by github-actions[bot]

ChristianTremblay, 2025-04-12:
Handled in develop branch