ChristianTremblay/BAC0 #516: Unable to subscript to COV: Error occurred: 'Lite' object has no attribute 'cov'

amaduain, 2025-01-21:
Using the latest release: 
```
BAC0              2024.9.20
bacpypes          0.18.7
bacpypes3         0.0.98
bokeh             3.6.2
```

When try to subscribe to COV, using the docs reference https://bac0.readthedocs.io/en/stable/cov.html

The bacnet object does not have the cov:

```
>>> import asyncio
>>> import asyncio
>>> import BAC0
>>> import yaml
>>> import pandas
>>> import bokeh
>>> bacnet = BAC0.start(ip='192.168.2.5/24')
2025-01-21 20:55:46,308 - INFO    | Starting Asynchronous BAC0 version 2024.09.10 (Lite)
2025-01-21 20:55:46,308 - INFO    | Using bacpypes3 version 0.0.98
2025-01-21 20:55:46,308 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2025-01-21 20:55:46,308 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2025-01-21 20:55:46,337 - INFO    | Using ip : 192.168.2.5/24 on port 47808 | broadcast : 192.168.2.255
2025-01-21 20:55:46,354 - INFO    | Using default JSON configuration file
2025-01-21 20:55:46,371 - INFO    | Registered as BACnet/IP App | mode normal
2025-01-21 20:55:46,374 - INFO    | Device instance (id) : 3056413
2025-01-21 20:55:46,375 - INFO    | Installing recurring task Ping Task (id:140718333628304)
>>> bacnet.discover()
>>> 2025-01-21 20:55:51,189 - WARNING | BAC0.scripts.Lite | Lite | BAC0.core.functions.Alias | Request timed out for what_is_network_number, no response
2025-01-21 20:55:55,192 - INFO    | Found those networks : set()
2025-01-21 20:55:55,192 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
2025-01-21 20:55:55,192 - INFO    | Issuing a local broadcast whois request.
2025-01-21 20:55:58,203 - INFO    | Discovery done. Found 1 devices on 0 BACnet networks.

>>>
>>> bacnet.cov
Traceback (most recent call last):
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3.11/asyncio/__main__.py", line 34, in callback
    coro = func()
           ^^^^^^
  File "<console>", line 1, in <module>
AttributeError: 'Lite' object has no attribute 'cov'
>>> bacnet.cov()
Traceback (most recent call last):
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3.11/asyncio/__main__.py", line 34, in callback
    coro = func()
           ^^^^^^
  File "<console>", line 1, in <module>

```

I'm trying to migrate from an older release and I need the COV with callback. 
Can you point me in the right direction? Is the documentation outdated, or is there another way?

Thxxxxx

ChristianTremblay, 2025-02-21:
I made cov work in the context of a BAC0.device "point"
`await cvm['ZN1-SP'].subscribe_cov()`
for example. So the value gets updated as soon as the value change.

`await cvm['ZN1-SP'].cancel_cov()` would cancel the subscription

I'm actually looking at adding support for a callback there. Eventually, I'll work on the COV function at the higher BAC0 level (bacnet.cov....)

ChristianTremblay, 2025-02-21:
If you ever have time, there is an attempt in develop

stillwelljt, 2025-02-26:
I'm having trouble getting this working using what's in the development branch. I can subscribe, but I never get the COV notification and the callback never gets called. This general thing worked in the pre asyncio version of BAC0. The gist of my code is:

```
cov_notification_event = threading.Event()

def my_callback(property_identifier, property_value):
   cov_notification_event.set()

bacnet.cov(controller.ip_address + ':47808', ('analogValue', 10), callback=my_callback)
cov_notification_event.wait()
cov_notification_event.clear()
```

Any ideas would be appreciated.

ChristianTremblay, 2025-02-26:
let start simple... can you just subscribe to COV without the callback ? 
Do you see the COV request on Wireshark ?

stillwelljt, 2025-02-28:
I did some additional poking around and got it working using the following to wait for the COV notification instead of using threading and .wait, .set, and .clear. I think we can chalk it up to the switch to asyncio. Thanks for the response!

```
while True:
   await asyncio.sleep(0)
```

ChristianTremblay, 2025-03-01:
Absolutely. Mixing async with thread will lead to a nightmare !

closed 2025-05-07 by github-actions[bot]