ChristianTremblay/BAC0 #520: Can't write floats/reals.

PeterSchulenberg, 2025-02-05:
https://github.com/ChristianTremblay/BAC0/blob/ef972e31577ebbcb3df6983ad69bbfbcd52b4a9c/BAC0/core/io/Write.py#L52C17-L52C287

Leads to an inability to write float/reals as their `.` will break the `\w`. Suggestion, change value matcher to `(?P<value>-*\w*(?:\.\w+)?)?` instead of `(?P<value>-*\w*)?`

CoMPaTech, 2025-02-05:
@ChristianTremblay I could write a small PR for this like the previous finding, just let us know

ChristianTremblay, 2025-02-09:
I'm pretty sure I worked on something related to that in the develop branch. Have you tried it ?

PeterSchulenberg, 2025-02-10:
> I'm pretty sure I worked on something related to that in the develop branch. Have you tried it ?
> 
> ```
> (?P<value>-?[\w\d]*\.?\d*)
> ```

@ChristianTremblay I just tested your proposal and it is also working. Thanks for the tip!

closed 2025-02-10 by PeterSchulenberg