ChristianTremblay/BAC0 #522: Read not working while using same port and different IP on Linux

kuriakosejohn, 2025-02-05:
The following test works fine on windows, but not on Linux. Looks like when there are multiple BAC0 instances (using same UDP port) on Linux environment, it is not working.

```python
import BAC0
import pytest
from BAC0.core.devices.local.factory import analog_input
from bacpypes3.basetypes import EngineeringUnits

@pytest.mark.asyncio
async def test_read_using_same_port():
    '''
    Test BAC0 API to read single property from BACnet objects
    '''
    # create BACnet device as server
    async with BAC0.start(ip="127.0.0.1/24") as server:
        # add objects
        ai_obj = analog_input(
            name="ZN-T",
            properties={"units": "degreesCelsius"},
            description="Zone Temperature",
            presentValue=21,
        )
        ai_obj.add_objects_to_application(server)

        # create BACnet client
        async with BAC0.start(ip="127.0.0.2/24") as client:
            present_value = await client.read('127.0.0.1 analogInput 0 presentValue')
            assert present_value == 21
            unit = await client.read('127.0.0.1 analogInput 0 units')
            assert unit == EngineeringUnits.degreesCelsius
```

ChristianTremblay, 2025-02-09:
This is probably more on the bacpypes3 side and the way the socket works on Linux vs Windows 

I don't use 127.0.0.1 usually and stick to the real(?) IP address of the interface. 

I also declare multiple IP address per interface if required.

e8ke, 2025-03-04:
Even though 127.0.0.1 and 127.0.0.2 seem to be different addresses, ultimatively they link to the one and only loop-back module of your OS.
E.g. I can not bind a server to 127.0.0.1:3000 and 127.0.0.2:3000 simultaneously. 
Had exactly the same issue... You could increment the port to separate server from client.

closed 2025-05-12 by github-actions[bot]