ChristianTremblay/BAC0 #523: RuntimeError: no running event loop sys:1: RuntimeWarning: coroutine 'Task.execute' was never awaited

Raphaeldufour, 2025-02-07:
Hello,
I keep getting this error. I know about python -m asyncio, but can't I just do a "python script.py" ?

closed 2025-02-07 by Raphaeldufour

ChristianTremblay, 2025-02-07:
Je suppose que tu t'es répondu :)

Voici un example que je viens d'utiliser dans une discussion

```python
import BAC0
import socket
import asyncio

async def start_bacnet_server():
    """
    Initialize the BACnet server and periodically update object values.
    """
    async with BAC0.start(deviceId=1112) as device1:
        # Start BACnet on a specific IP and port

        print("BACnet started. Discovering devices...")

        # Discover devices (this may take some time)
        await device1._discover()
        # Wait a few seconds to allow device discovery to complete
        #await asyncio.sleep(3)

        # Get the discovered devices
        devices = await device1.devices
        print(f"Discovered devices: {devices}")

if __name__ == "__main__":
    asyncio.run(start_bacnet_server())
```

Raphaeldufour, 2025-02-07:
Salut merci de ta réponse, par contre j'ai un problème, le discovery trouve bien les devices mais quand je print device1.devices cela met Discovered devices: None
2025-02-07 16:55:28,255 - INFO    | Installing recurring task Ping Task (id:1487325253648)
2025-02-07 16:55:32,836 - INFO    | Found those networks : {1}
2025-02-07 16:55:32,837 - INFO    | Discovering network 1
2025-02-07 16:55:35,847 - INFO    | Discovery done. Found 3 devices on 1 BACnet networks.
Discovered devices: None
2025-02-07 16:55:55,866 - INFO    | Stopping all tasks
2025-02-07 16:55:55,866 - INFO    | Ok all tasks stopped
2025-02-07 16:55:55,867 - INFO    | BACnet stopped
2025-02-07 16:55:55,867 - INFO    | BAC0|3056388 disconnected. Exiting context manager.

`import BAC0
import socket
import asyncio

async def start_bacnet_server():
    """
    Initialize the BACnet server and periodically update object values.
    """
    async with BAC0.start(ip='10.79.9.84') as device1:
        # Start BACnet on a specific IP and port

        print("BACnet started. Discovering devices...")

        # Discover devices (this may take some time)
        await device1._discover()
        # Wait a few seconds to allow device discovery to complete
        await asyncio.sleep(20)

        # Get the discovered devices
        devices = await device1.devices
        print(f"Discovered devices: {devices}")

if __name__ == "__main__":
    asyncio.run(start_bacnet_server())`

ChristianTremblay, 2025-02-07:
J'avais oublié.... bacnet.devices rretourne pas les valeurs....

```python
import BAC0
import socket
import asyncio

# Retrieving the IP address from the host PC
hostname = socket.gethostname()
host_ip = socket.gethostbyname(hostname)  # Get the corresponding IP address
print(host_ip)

async def start_bacnet_server():
    """
    Initialize the BACnet server and periodically update object values.
    """
    async with BAC0.start(ip="10.138.103.17/16", deviceId=1112) as device1:
        # Start BACnet on a specific IP and port
        #device1 = BAC0.start(ip=host_ip, port=47811, deviceId=1112)

        print("BACnet started. Discovering devices...")

        # Discover devices (this may take some time)
        await device1._discover()
        # Wait a few seconds to allow device discovery to complete
        #await asyncio.sleep(3)

        # Get the discovered devices
        devices = device1.discoveredDevices
        print(f"Discovered devices: {devices}")

        print("Pretty")
        await device1.devices

if __name__ == "__main__":
    asyncio.run(start_bacnet_server())
```

ChristianTremblay, 2025-02-11:
J'oriente la lecture des trendlogs à partir d'un `BAC0.device`

Y'a encore un peu de travail pour être capable de facilement les retrouver... mais j'ai une exemple ici qui pourrait t'aider.

```python
import BAC0
bacnet = BAC0.start(ip='192.168.142.20/24')
await bacnet.use_router(('192.168.142.1:47808', 65001))
salle = await BAC0.device('65001:7', 5207, bacnet)

a = (await salle.trendlogs[0].history, salle.trendlogs[0].properties.name)
b = (await salle.trendlogs[1].history, salle.trendlogs[1].properties.name)
c = (await salle.trendlogs[2].history, salle.trendlogs[2].properties.name)
d = (await salle.trendlogs[3].history, salle.trendlogs[3].properties.name)
e = (await salle.trendlogs[4].history, salle.trendlogs[4].properties.name)
f = (await salle.trendlogs[5].history, salle.trendlogs[5].properties.name)
g = (await salle.trendlogs[6].history, salle.trendlogs[6].properties.name)
h = [a,b,c,d,e,f,g]
import pandas as pd
df = pd.DataFrame()
for each in h:
    his, name = each
    df.insert(0, name, his)
    #his.save(f'{name}.csv')
df.to_excel('all.xlsx')
```