ChristianTremblay/BAC0 #526: Module 'BAC0' has no attribute 'start' (HomeAssistant)

Michelepooh, 2025-02-13:
I used to use a custom integration in HomeAssistant, and it worked very well until HA switched to async. 
Now, I cannot start BAC0: BAC0.connect(), BAC0.start(), BAC0.lite() – for all of them I'm getting the same error message "Error initializing BAC0: module 'BAC0' has no attribute ...".
The only attribute that does not give this error is BAC0.device() but it doesn't work anyway.
What do I do wrong?

(python 3.13, BAC0 24.09.20, bacpypes3 0.0.97, HA 2025 Raspberry installation; all code rewritten for async)

Here is my __init.py__ from the custom integration:

import BAC0
import logging
from importlib.metadata import version
from homeassistant.config_entries import ConfigEntry
from homeassistant.core import HomeAssistant

DOMAIN = "home_MK"
_LOGGER = logging.getLogger(__name__)

class BacnetRunner:
    def __init__(self, ip):
        self.ip = ip
        self.bacnet = None

    async def initialize(self, hass: HomeAssistant):
        if self.bacnet is None:
            try:
                self.bacnet = await BAC0.start(self.ip)
            except Exception as e:
                _LOGGER.error(f"Error initializing BAC0: {e}")
                raise

    async def _read_command(self, command):
        command_str = " ".join(command)
        return await self.bacnet.read(
            command_str, vendor_id=1111
        )

    async def _write_command(self, command):
        command_str = " ".join(command)
        await self.bacnet.write(command_str, vendor_id=1111)

    async def read_object_value(self, prefix, prop):
        command = [prefix, prop]
        return await self._read_command(command)

    async def read_present_value(self, prefix):
        command = [prefix, "presentValue"]
        return await self._read_command(command)

    async def read_dimmer_value(self, prefix):
        command = [prefix, "530"]
        return await self._read_command(command)

    async def set_object_value(self, prefix, prop, value):
        command = [prefix, prop, str(value)]
        await self._write_command(command)

    async def set_bin_value(self, prefix, *, active=True):
        if active:
            value = "active"
        else:
            value = "inactive"
        await self.set_object_value(prefix, "presentValue", value)

    async def set_dimmer_value(self, prefix, *, brightness):
        await self.set_object_value(prefix, "530", str(brightness))

async def async_setup_entry(hass: HomeAssistant, entry: ConfigEntry) -> bool:
    local_ip = entry.data.get("local_ip")
    bacnet_runner = BacnetRunner(local_ip)
    await bacnet_runner.initialize(hass)

    hass.data.setdefault(DOMAIN, {})
    hass.data[DOMAIN]["bacnet_runner"] = bacnet_runner

    await hass.config_entries.async_forward_entry_setups(entry, ["light", "sensor", "button"])
    return True

async def async_unload_entry(hass: HomeAssistant, entry: ConfigEntry) -> bool:
    """Unload a config entry."""
    unload_ok = await hass.config_entries.async_unload_platforms(entry, ["light", "sensor", "button"])
    if unload_ok:
        hass.data.pop(DOMAIN, None)
    return unload_ok

ChristianTremblay, 2025-02-14:
Could it be an old version of BAC0 installed in the environment running HA ?

I'm not well versed in HA, so I'll take a moment to look at your code

Michelepooh, 2025-02-14:
> Could it be an old version of BAC0 installed in the environment running HA ?

<details>
  <summary>Thanks!
The "version" command says "2024.09.20", however in the error messages I see "2024.09.10". 
I also checked inspect.getmembers(BAC0) and didn't see neither "start" nor "connect" there. 

_(the reply to inspect.getmembers(BAC0) is under the cut)_ </summary>
  
Available methods in BAC0: [('Base', <class 'BAC0.scripts.Base.Base'>), ('TrendLog', <function TrendLog at 0x7f7c3ac9a0>), ('__builtins__', {'__name__': 'builtins', '__doc__': "Built-in functions, types, exceptions, and other objects.\n\nThis module provides direct access to all 'built-in'\nidentifiers of Python; for example, builtins.len is\nthe full name for the built-in function len().\n\nThis module is not normally accessed explicitly by most\napplications, but can be useful in modules that provide\nobjects with the same name as a built-in value, but in\nwhich the built-in of that name is also needed.", '__package__': '', '__loader__': <class '_frozen_importlib.BuiltinImporter'>, '__spec__': ModuleSpec(name='builtins', loader=<class '_frozen_importlib.BuiltinImporter'>, origin='built-in'), '__build_class__': <built-in function __build_class__>, '__import__': <built-in function __import__>, 'abs': <built-in function abs>, 'all': <built-in function all>, 'any': <built-in function any>, 'ascii': <built-in function ascii>, 'bin': <built-in function bin>, 'breakpoint': <built-in function breakpoint>, 'callable': <built-in function callable>, 'chr': <built-in function chr>, 'compile': <built-in function compile>, 'delattr': <built-in function delattr>, 'dir': <built-in function dir>, 'divmod': <built-in function divmod>, 'eval': <built-in function eval>, 'exec': <built-in function exec>, 'format': <built-in function format>, 'getattr': <built-in function getattr>, 'globals': <built-in function globals>, 'hasattr': <built-in function hasattr>, 'hash': <built-in function hash>, 'hex': <built-in function hex>, 'id': <built-in function id>, 'input': <built-in function input>, 'isinstance': <built-in function isinstance>, 'issubclass': <built-in function issubclass>, 'iter': <built-in function iter>, 'aiter': <built-in function aiter>, 'len': <built-in function len>, 'locals': <built-in function locals>, 'max': <built-in function max>, 'min': <built-in function min>, 'next': <built-in function next>, 'anext': <built-in function anext>, 'oct': <built-in function oct>, 'ord': <built-in function ord>, 'pow': <built-in function pow>, 'print': <built-in function print>, 'repr': <built-in function repr>, 'round': <built-in function round>, 'setattr': <built-in function setattr>, 'sorted': <built-in function sorted>, 'sum': <built-in function sum>, 'vars': <built-in function vars>, 'None': None, 'Ellipsis': Ellipsis, 'NotImplemented': NotImplemented, 'False': False, 'True': True, 'bool': <class 'bool'>, 'memoryview': <class 'memoryview'>, 'bytearray': <class 'bytearray'>, 'bytes': <class 'bytes'>, 'classmethod': <class 'classmethod'>, 'complex': <class 'complex'>, 'dict': <class 'dict'>, 'enumerate': <class 'enumerate'>, 'filter': <class 'filter'>, 'float': <class 'float'>, 'frozenset': <class 'frozenset'>, 'property': <class 'property'>, 'int': <class 'int'>, 'list': <class 'list'>, 'map': <class 'map'>, 'object': <class 'object'>, 'range': <class 'range'>, 'reversed': <class 'reversed'>, 'set': <class 'set'>, 'slice': <class 'slice'>, 'staticmethod': <class 'staticmethod'>, 'str': <class 'str'>, 'super': <class 'super'>, 'tuple': <class 'tuple'>, 'type': <class 'type'>, 'zip': <class 'zip'>, '__debug__': True, 'BaseException': <class 'BaseException'>, 'BaseExceptionGroup': <class 'BaseExceptionGroup'>, 'Exception': <class 'Exception'>, 'GeneratorExit': <class 'GeneratorExit'>, 'KeyboardInterrupt': <class 'KeyboardInterrupt'>, 'SystemExit': <class 'SystemExit'>, 'ArithmeticError': <class 'ArithmeticError'>, 'AssertionError': <class 'AssertionError'>, 'AttributeError': <class 'AttributeError'>, 'BufferError': <class 'BufferError'>, 'EOFError': <class 'EOFError'>, 'ImportError': <class 'ImportError'>, 'LookupError': <class 'LookupError'>, 'MemoryError': <class 'MemoryError'>, 'NameError': <class 'NameError'>, 'OSError': <class 'OSError'>, 'ReferenceError': <class 'ReferenceError'>, 'RuntimeError': <class 'RuntimeError'>, 'StopAsyncIteration': <class 'StopAsyncIteration'>, 'StopIteration': <class 'StopIteration'>, 'SyntaxError': <class 'SyntaxError'>, 'SystemError': <class 'SystemError'>, 'TypeError': <class 'TypeError'>, 'ValueError': <class 'ValueError'>, 'Warning': <class 'Warning'>, 'FloatingPointError': <class 'FloatingPointError'>, 'OverflowError': <class 'OverflowError'>, 'ZeroDivisionError': <class 'ZeroDivisionError'>, 'BytesWarning': <class 'BytesWarning'>, 'DeprecationWarning': <class 'DeprecationWarning'>, 'EncodingWarning': <class 'EncodingWarning'>, 'FutureWarning': <class 'FutureWarning'>, 'ImportWarning': <class 'ImportWarning'>, 'PendingDeprecationWarning': <class 'PendingDeprecationWarning'>, 'ResourceWarning': <class 'ResourceWarning'>, 'RuntimeWarning': <class 'RuntimeWarning'>, 'SyntaxWarning': <class 'SyntaxWarning'>, 'UnicodeWarning': <class 'UnicodeWarning'>, 'UserWarning': <class 'UserWarning'>, 'BlockingIOError': <class 'BlockingIOError'>, 'ChildProcessError': <class 'ChildProcessError'>, 'ConnectionError': <class 'ConnectionError'>, 'FileExistsError': <class 'FileExistsError'>, 'FileNotFoundError': <class 'FileNotFoundError'> [... cut ...] python3.13/site-packages/BAC0/infos.py'>), ('load', <class 'BAC0.core.devices.Device.DeviceLoad'>), ('load_dotenv', <function load_dotenv at 0x7f7fc7bf60>), ('log_level', <function update_log_level at 0x7f7c342de0>), ('os', <module 'os' (frozen)>), ('scripts', <module 'BAC0.scripts' from '/usr/local/lib/python3.13/site-packages/BAC0/scripts/__init__.py'>), ('tasks', <module 'BAC0.tasks' from '/usr/local/lib/python3.13/site-packages/BAC0/tasks/__init__.py'>), ('version', '2024.09.10')]
</details>

ChristianTremblay, 2025-02-14:
Hummm... `await BAC0.start` is incorrect I think.

start is a shortcut for the class "Lite" (https://github.com/ChristianTremblay/BAC0/blob/develop/BAC0/__init__.py#L39)

Try removing "await"

Michelepooh, 2025-02-15:
Resolved!
Home Assistant had an older version of influxdb-client library (1.24, not async). 
Updating this library manually helped. 
Thanks, Christian!

closed 2025-02-15 by Michelepooh

ChristianTremblay, 2025-02-15:
Haaa ! Nice ! Good catch !

Do you think an official BAC0 support in HA is required ?

Michelepooh, 2025-02-16:
> Do you think an official BAC0 support in HA is required ?

I think that yes, it would be great – HA's popularity grows and I'm sure there are plenty of BACNet devices around that could be connected to HA.