ChristianTremblay/BAC0 #530: Can I add a way to get all the information about the location device if it has any information, and the write method doesn't work in the latest version

Rooike111, 2025-02-20:
Can I add a way to get all the information about the location device if it has any information, and the write method doesn't work in the latest version

KonstantinMasich, 2025-02-20:
@Rooike111  Use `_write` instead of `write` for now

Issue about the write method:

https://github.com/ChristianTremblay/BAC0/issues/498

Rooike111, 2025-02-21:
Hello, thank you for your kind reply, but I still want to know. Like java's BACnet4j library, all I need to know is the address of the Bacnet device and I can query all the information on that device. BAC0, on the other hand, only when I know what types the device supports and how many of each type there are can I fully read all of its information. Can you read all his information at once, without me knowing anything about the device at all? Thank you very much.

KonstantinMasich, 2025-02-21:
@Rooike111 well, I used `bacnet.discover(...)` and `bacnet.who_is(...)`, and also `bacnet.devices` to inspect the devices. With `device.bacnet_properties` you can see the device properties; that's probably what you wanted?

ChristianTremblay, 2025-02-21:
Creating a BAC0.device does a lot of the heavy lifting

closed 2025-04-30 by github-actions[bot]