ChristianTremblay/BAC0 #533: BAC0 Device Connection Error

rtarpara, 2025-03-12:
Hi,
I am trying to establish connection with one of the BACnet MSTP device and discover the points. I am getting below error. I have attached the script and complete error. Can you please help and let me know the issue?

'BAC0.core.io.IOExceptions.BadDeviceDefinition: Please provide address, device id and network or specify from_backup argument'

[BACnet_Device_Read_Script.txt](https://github.com/user-attachments/files/19201054/BACnet_Device_Read_Script.txt)
[BACnet_Device_Read_Script_Error.txt](https://github.com/user-attachments/files/19201055/BACnet_Device_Read_Script_Error.txt)

ChristianTremblay, 2025-03-15:
```python
import asyncio
import BAC0

bacnet = None

async def read_bacnet_device(address, device_id, network):
    async with BAC0.start(ip="192.168.201.5/24") as bacnet:
        my_device = await BAC0.device(address, device_id , bacnet)
        print(my_device.points)

async def main():
    # Create task for running coroutines concurrently
    # Replace with your actual BACnet device address and  details

    await read_bacnet_device('17251:1', 1725101)

if __name__ == "__main__":
    asyncio.run(main())
```

rtarpara, 2025-03-15:
Thanks Christian for correcting the script and sending me it back! I tried
below. It seems working and generating points but somehow it is not
printing. This device has few BACnet points, and I have confirmed using Yet
Another BACnet Explorer. Please see snapshot. Can you please check?
Does point save automatically in sqlite3 database? How do I find
table name? I am learning Python. Sorry for too many questions!

[image: image.png]

import asyncio
import BAC0

bacnet = None

async def read_bacnet_device(address, device_id):
    async with BAC0.start(ip="192.168.201.5/24") as bacnet:
        my_device = await BAC0.device(address, device_id , bacnet)
        print (my_device.points)

async def main():
    # Create task for running coroutines concurrently
    # Replace with your actual BACnet device address and  details

    await read_bacnet_device('192.168.201.4', 2532100)

if __name__ == "__main__":
    asyncio.run(main())

C:\Users\User1\PycharmProjects\BACnet\.venv\Scripts\python.exe
C:\Users\User1\PycharmProjects\BACnet\Test1.py
2025-03-15 12:43:21,295 - INFO    | Starting Asynchronous BAC0 version
2024.09.10 (Lite)
2025-03-15 12:43:21,295 - INFO    | Using bacpypes3 version 0.0.99
2025-03-15 12:43:21,295 - INFO    | Use BAC0.log_level to adjust verbosity
of the app.
2025-03-15 12:43:21,296 - INFO    | Ex. BAC0.log_level('silence') or
BAC0.log_level('error')
2025-03-15 12:43:21,495 - INFO    | Using ip : 192.168.201.5/24 on port
47808 | broadcast : 192.168.201.255
2025-03-15 12:43:21,580 - INFO    | Using default JSON configuration file
2025-03-15 12:43:21,591 - INFO    | Registered as BACnet/IP App | mode
normal
2025-03-15 12:43:21,593 - INFO    | Device instance (id) : 3056851
2025-03-15 12:43:21,593 - INFO    | BAC0|3056851 connected. Entering
context manager.
2025-03-15 12:43:21,596 - INFO    | Changing 2532100 state to
DeviceDisconnected'>
2025-03-15 12:43:21,599 - INFO    | Installing recurring task Ping Task
(id:2447250903392)
2025-03-15 12:43:21,629 - INFO    | Changing 2532100 state to
RPMDeviceConnected'>
2025-03-15 12:43:21,660 - INFO    | Device 2532100:[BACnetExport_2532100]
found... building points list
2025-03-15 12:43:21,849 - INFO    | Points and trendlogs (if any) created
2025-03-15 12:43:21,849 - INFO    | Device defined for normal polling with
a delay of 10sec
2025-03-15 12:43:21,849 - INFO    | Polling started, values read every 10
seconds
2025-03-15 12:43:21,849 - INFO    | Device BACnetExport_2532100 | 2532100
ready, use device_name.points and start interact with it
2025-03-15 12:43:21,853 - INFO    | Wait while stopping polling for
BACnetExport_2532100
2025-03-15 12:43:21,853 - INFO    | BACnetExport_2532100 | Polling stopped
2025-03-15 12:43:21,853 - INFO    | Saving BACnetExport_2532100 to
database...
2025-03-15 12:43:21,859 - ERROR   | BAC0.core.devices.Device | Device |
BAC0.db.sql | Impossible to save right now, error in data
2025-03-15 12:43:21,862 - ERROR   | Error saving to SQL database
2025-03-15 12:43:21,865 - INFO    | Device saved to Device_2532100.db
2025-03-15 12:43:21,865 - INFO    | Changing 2532100 state to DeviceFromDB'>
2025-03-15 12:43:21,865 - INFO    | Initializing DB for BACnetExport_2532100
2025-03-15 12:43:21,872 - WARNING | No history retrieved from
Device_2532100.db:
2025-03-15 12:43:21,872 - INFO    | BACnetExport_2532100 restored from db
2025-03-15 12:43:21,873 - INFO    | You can reconnect to network using :
"device.connect(network=bacnet)"
2025-03-15 12:43:21,875 - INFO    | Stopping all tasks
2025-03-15 12:43:21,875 - INFO    | Ok all tasks stopped
2025-03-15 12:43:21,877 - INFO    | BACnet stopped
2025-03-15 12:43:21,877 - INFO    | BAC0|3056851 disconnected. Exiting
context manager.
Traceback (most recent call last):
  File "C:\Users\User1\PycharmProjects\BACnet\Test1.py", line 19, in
<module>
    asyncio.run(main())
    ~~~~~~~~~~~^^^^^^^^
  File "C:\Program
Files\WindowsApps\PythonSoftwareFoundation.Python.3.13_3.13.752.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py",
line 195, in run
    return runner.run(main)
           ~~~~~~~~~~^^^^^^
  File "C:\Program
Files\WindowsApps\PythonSoftwareFoundation.Python.3.13_3.13.752.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py",
line 118, in run
    return self._loop.run_until_complete(task)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "C:\Program
Files\WindowsApps\PythonSoftwareFoundation.Python.3.13_3.13.752.0_x64__qbz5n2kfra8p0\Lib\asyncio\base_events.py",
line 725, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "C:\Users\User1\PycharmProjects\BACnet\Test1.py", line 16, in main
    await read_bacnet_device('192.168.201.4', 2532100)
  File "C:\Users\User1\PycharmProjects\BACnet\Test1.py", line 9, in
read_bacnet_device
    print (my_device.points)
    ~~~~~~^^^^^^^^^^^^^^^^^^
  File
"C:\Users\User1\PycharmProjects\BACnet\.venv\Lib\site-packages\BAC0\core\devices\Points.py",
line 782, in __repr__
    return f"{self.properties.device.properties.name}/{self.properties.name}
: {val:.2f} {self.properties.units_state}"

   ^^^^^^^^^
ValueError: Unknown format code 'f' for object of type 'str'

Process finished with exit code 1

ChristianTremblay, 2025-03-17:
I suspect there was not yet a value retrieved and the application considered the string 'NaN' in the `__repr__` function. I will try to fix this.

I also suspect the script end up to fast... I strongly suggest you try Jupyter Notebook to explore. You will be able to keep the script running and explore points, and different functions. When running a `.ipynb` file in VSCode for example, it is very easy, the event loop already runs so you just have to define `bacnet = BAC0.start()` and you're done.

For this particular script, you could add some kind of delay before ending the script... 

```python
import asyncio
import BAC0

bacnet = None

async def read_bacnet_device(address, device_id):
    async with BAC0.start(ip="192.168.201.5/24") as bacnet:
        my_device = await BAC0.device(address, device_id , bacnet)
        print (my_device.points)
        await asyncio.sleep(1)

async def main():
    # Create task for running coroutines concurrently
    # Replace with your actual BACnet device address and  details

    await read_bacnet_device('192.168.201.4', 2532100)

if __name__ == "__main__":
    asyncio.run(main())
```

rtarpara, 2025-03-17:
Thanks. I will try in Jupiter.

closed 2025-05-25 by github-actions[bot]