ChristianTremblay/BAC0 #539: Initiate device with object_list

majuss, 2025-03-28:
I try to initiate a controller with a object_list. I found inconsistencies, sometimes its called object**s**_list.

I do the following, but the property is still empty.
```python
import asyncio
import BAC0
from bacpypes3.object import AnalogInputObject

bacnet = None

analog_input = [AnalogInputObject(
    objectName='Temperature Sensor',
    presentValue=23.5,
    units='degreesCelsius'
)]

async def main():
    # Create task for running coroutines concurrently
    # Replace with your actual BACnet device address and  details

    async with BAC0.start(ip="192.168.1.216/24") as bacnet:
        my_device = await BAC0.device(object_list = analog_input, network=bacnet, device_id=1725101, address ='17251:1' )
        print(my_device.properties)

if __name__ == "__main__":
    asyncio.run(main())
```

ChristianTremblay, 2025-04-12:
object list is to create the local "image" of a remote controller. Not to create a local object in the running instance of BAC0. From the definition of your list, I suspect you mixed both concepts.

majuss, 2025-04-14:
Ah okay, could you please provide a complete minimal runnable example how I can initiate the controller with the object_lsit?

Because when I use you example object_list it also is empty:
```python
import asyncio
import BAC0
from bacpypes3.object import AnalogInputObject

bacnet = None

my_obj_list = [('file', 1),
             ('analogInput', 2),
             ('analogInput', 3),
             ('analogInput', 5),
             ('analogInput', 4),
             ('analogInput', 0),
             ('analogInput', 1)]

async def main():
    # Create task for running coroutines concurrently
    # Replace with your actual BACnet device address and  details

    async with BAC0.start(ip="192.168.1.133/24") as bacnet:
        my_device = await BAC0.device(object_list = my_obj_list, network=bacnet, device_id=1725101, address ='17251:1' )
        print(my_device.properties)

if __name__ == "__main__":
    asyncio.run(main())
```

majuss, 2025-05-06:
@ChristianTremblay Hi Christian, I also looked at bacpypes3 directly, but the documentation is really sparse and I don't really understand the concept behind the library. I would be very happy about a minimal example.

leanfm, 2025-05-07:
In order to simulate a BACnet device, you don't need to use the device class, which is meant to connect with other devices on the network. You can create a BAC0 object and then create points into it. This page is the right example https://bac0.readthedocs.io/en/latest/local_objects.html

majuss, 2025-05-12:
Thank you @leanfm ! But unfortunately the doc example is missing the code: https://bac0.readthedocs.io/en/latest/local_objects.html#models "# code here". 

Furthermore most of the "samples" are outdated and won't run with the bacpypes3 version of the lib, for example: https://github.com/ChristianTremblay/BAC0/blob/main/samples/excel_from_discovered_devices.py

So I want to create a local device with objects and reat to who_is requests from the network. Is there a way to do this?

ChristianTremblay, 2025-05-14:
Try this : https://github.com/ChristianTremblay/BAC0/blob/main/tests/manual_test_create_device.py

closed 2025-07-21 by github-actions[bot]