ChristianTremblay/BAC0 #542: BAC0.core.io.IOExceptions.UnknownPropertyError part 2

pisuke, 2025-05-12:
Hi Christian,

> I'm not sure if I should just return "noUnit" or return `EngineeringUnits("noUnits")` like I did, so if you can test, that would help me.
> 
> Thanks 

 _Originally posted by @ChristianTremblay in [#352](https://github.com/ChristianTremblay/BAC0/issues/352#issuecomment-2597307182)_

Thanks for providing a fix in https://github.com/ChristianTremblay/BAC0/issues/352#issuecomment-2597307182

I tried pulling https://github.com/ChristianTremblay/BAC0/commit/84a7e789d11e7ef0b3d0a23ae22d737e62fc219b in my script however now I get this error.

```
2025-05-12 10:52:41,021 - INFO    | Starting Asynchronous BAC0 version 2024.09.10 (Lite)
2025-05-12 10:52:41,021 - INFO    | Using bacpypes3 version 0.0.102
2025-05-12 10:52:41,021 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2025-05-12 10:52:41,021 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
Traceback (most recent call last):
  File "bacnet-scan.py", line 346, in <module>
  File "bacnet-scan.py", line 306, in main
  File "BAC0/scripts/Lite.py", line 141, in __init__
  File "BAC0/tasks/TaskManager.py", line 143, in start
  File "asyncio/tasks.py", line 417, in create_task
RuntimeError: no running event loop
[PYI-1761440:ERROR] Failed to execute script 'bacnet-scan' due to unhandled exception!
sys:1: RuntimeWarning: coroutine 'Task.execute' was never awaited
```

Sorry for leaving it so much time before testing, do you have any comments on what I can change to make it work?

Thank you,

Francesco

ChristianTremblay, 2025-05-14:
When running BAC0 now, you have to deal with asyncio.

One easy way is to start python using

`python -m asyncio`

and then you can import BAC0 and do your stuff....

For scripts, you can see an example [here](https://github.com/ChristianTremblay/BAC0/blob/main/tests/manual_test_create_device.py)

pisuke, 2025-05-15:
Thank you Christian, now I understand about the changes moving to bacpypes3. I'll follow your steps.

In the meantime, I've sorted my issues by forking BAC0 and applying some small patches to the 2023 code, see my changes here: https://github.com/pisuke/BAC0/compare/master...unknown-units-2023 
With these changes, I was able to complete BACnet scans in several sites where we have "bad" devices.