ChristianTremblay/BAC0 #548: Can't read BACnet value directly when device is known (BAC0 fails but BACpypes3/Node-RED succeed)

billybboy, 2025-06-20:
I'm attempting to read a BACnet value directly from a known device IP (192.168.226.191) using BAC0 v2025.06.10 (Lite), but the request consistently fails with `NoResponseFromController`. However, the same operation succeeds using bacpypes3 or Node-RED with identical device configuration .

- Environment
BAC0 Version: 2025.06.10
Python Version: 3.10
Host IP/Subnet: 192.168.225.36/24
Device IP (target): 192.168.226.191
Tested on: Linux

```
import BAC0
import asyncio

address = '192.168.226.191'
host_ip = "192.168.225.36"

async def read_values():
    async with BAC0.start() as b:
        print("BACnet started. Reading devices...")
        res = await b.read(f"{address} analog-value 100151 presentValue")
        print(res)

if __name__ == "__main__":
    asyncio.run(read_values())
```

[image]

[image]

it worked using bacpype3:

[image]
[image]

Am I doing something incorrectly?

ChristianTremblay, 2025-06-26:
Somewhere, you chose to use port 47810... 
can you use 47808 ?

ChristianTremblay, 2025-06-27:
Your problem is related to the nature of broadcast messages in BACnet.
Your instance of BAC0 is in a different subnet than the controller you are pointing to. 

The part of code you removed tries to retrieve information from the controller (target) using who_is. This command is sent using (probably 192.168.225.255, based on your IP/24). This command will not traverse to the other subnet (192.168.226.0/24) without help (often, a BBMD in BACnet).

I'm using info_cache in BAC0.device and I'm relying on the basis that the BACnet network should be configured properly. In this particular case, I understand that it may be annoying.

I'll have a further look at this. Looking up closely to your first screen shot, I also see that the script did make a unicast who_is call to the device... I suspect the answer was a broadcast (maybe an old device, unable to respond with a unicast IAM)...

billybboy, 2025-06-30:
Thanks a lot!

ChristianTremblay, 2025-11-01:
> I am developing a bacnet connector and planning to deploy on docker. Do you think this will be an issue. I am also facing issue with developing on wsl2 given how udp is not being broadcasted properly on wsl2

I know there have been work done by Joel on docker. I also know that you can be very flexible with docker networks.... so I would say there will be challenges, but eveything is possible they say.

You may need some BBMD mechanism...  Please let me know how you progress