ChristianTremblay/BAC0 #551: COV callback

pmolke, 2025-07-19:
As a test, I subscribed to a value via COV, as described in the documentation. Unfortunately, it doesn't work because the callback function is incorrect.

The documentation says def callback(elements)

I get the following error:
2025-07-19 17:53:30,145 - ERROR | Error in COV subscription : my_callback() got an unexpected keyword argument 'property_identifier'

I find this statement in the code:
 """
  Subscribe to COV notification for a given address and objectID
  If callback is provided, it will be called with the value of the COV notification
  So the function must be built to accept two arguments : property_identifier and property_value
  """
Doesn't this contradict the documentation? What am I doing wrong?

ChristianTremblay, 2025-08-14:
It is next on my list. Thanks for your patience.

ChristianTremblay, 2025-08-25:
https://github.com/ChristianTremblay/BAC0/blob/develop/tests/manual_test_cov.py show me the actual develop branch deals correctly with COV and callback.

I'm planning to merge into main soon.

closed 2025-08-25 by ChristianTremblay